ARMmbed/DAPLink #727: MAX32625PICO not listed here

Gerriko, 2020-08-26:
I went to https://armmbed.github.io/DAPLink/ in search for latest DAPLink firmware for the MAX32625PICO.

The only board found was MAX32625, which does not work.

Through trial and error I discovered that the MAX32625PICO bin files are in fact stored elsewhere in the MBED.com labyrinth, which is a not very useful location (IMHO).

https://os.mbed.com/teams/MaximIntegrated/wiki/MAX32625PICO-Firmware-Updates

Maybe you could update the github DAPLink search facility accordingly.

0xc0170, 2020-11-02:
Thanks for reporting.

My understanding is that daplink page searches only the releases from this repository (aka official releases). There are many more (even from private forks) available on os.mbed.com

mathias-arm, 2021-10-06:
The images are now at https://github.com/MaximIntegratedMicros/max32625pico-firmware-images

ozersa, 2023-02-03:
> uld update the github DAPLink search facility accordingly.

Update, it is here:
https://github.com/MaximIntegrated/max32625pico-firmware-images/