ARMmbed/DAPLink #729: Add support for signed binary files

timdesi, 2020-09-10:
I have FRDM_K64F development board with latest available version of DAPLink. Need to install signed firmware with drag and drop. My problem is that signed .bin firmware files filed with Error below and this crashes board, from crashing i mean to recover board i need install bootloader again.

```
FAIL.TXT
error: The transfer timed out.
type: transient, user 

```

I am signing firmwares with https://mcuboot.com/mcuboot/imgtool.html.

PYOCD console tool also works without problems.
`pyocd flash -e sector -a 0x20000 -t k64f signed.bin`

Also signed .hex firmware files are installed without problems.

DETAILS.TXT
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0240000048824e45000c7007dd89001e8761000097969900
HIC ID: 97969900
Auto Reset: 0
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0253
Bootloader Version: 0244
Git SHA: 64359f5c786363065a41ec15c348e3d53568da03
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0xca4adf2f
Interface CRC: 0x08858a9a
Remount count: 4
URL: https://mbed.org/device/?code=@U?version=@V?target_id=@T
```

This issue looks similar.
#354

flit, 2020-09-10:
The problem is that binary files do not have an associated base address, so they are written to the start of flash. The solution is to use hex files (as you note, they work) that have embedded addresses.

timdesi, 2020-09-11:
Thx. @flit for you kind answer.

closed 2020-09-11 by timdesi