ARMmbed/DAPLink #732: daplink firmware for nrf52840 does not detect external target

joicetm, 2020-09-30:
Hi,

i am using mbed-studio to develop firmware for my custom board based on nrf52840 but i am not able to upload the binaries to my board using nrf52840 dk with daplink software.

As i can see, it is able to flash the binaries to the dk but it does not detect external target. ie.. It is not able to upload the binaries from mbed studio to my custom board through nrf52840 dk. i have connected  my custom board to the swd lines of nrf52840 dk. 

The jlink software that comes pre-loaded with nrf5240 dk is able to detect external target if present. it is done based on checking the vtg pin level.

It would be very good if the daplink software can detect the external target connected on an nrf52840 dk.

cederom, 2020-09-30:
1. Are you sure the local onboard Target (nRF52840) is disconnected / powered off so it does not interfere with SWD bus?

2. Have you tried pyOCD and its flashtool? It should allow you to work with other Targets with your DAPLink over CMSIS-DAP endpoint.

3. DAPLink has hardcoded FlashAlgo for Drag-n-Drop functionality for a specific Target (nRF52840 in this case). It will work only with this specific hardcoded Target with Drag-n-Drop functionality.

joicetm, 2020-10-01:
Hi @cederom 

Thanks for the reply.

_Are you sure you are using DAPLink firmware and you have latest version?_

yes, i have 0253_sam3u2c_mkit_dk_dongle_nrf5x_0x5000.bin

2 _Are you sure the local onboard Target (nRF52840) is disconnected_

There is no easy way to turn off the onboard target, the only way i know is to cut the trace.
fyi,It works fine while using it with jlink software. no need to turn off the onboard target. (i think the software  has the ability to choose the target based on the voltage level of vtg pin on the dk)

3 _Have you tried pyOCD and its flashtool?_

No, surely i will give it a try and let you know. what i am trying to do is to make the process simple, ie.. to program it by simply clicking the upload button on the mbed-studio.
I am able to upload the  mbed compiled binaries by using nrfjprog, i have selected the target as jlnk and built hex files from mbed-studio and used a script to upload the binaries with nrfjprog.(using dk as jlink programmer)

I know there are several ways to upload to external target like by using external programmers. since most of the nrf developers use dk as programmer to upload to custom target, i thought it would be a good addition if we can use dk as programmer to upload the code to external target while working with mbed studio.

cederom, 2020-10-01:
with pyOCD you can flash and debug any external target. just keep the internal Target powered down. 

use pyOCD and let us know how that works - you can simply use Python virtual environment for that (pip install pyocd) :-)

mbrossard, 2021-09-10:
Looking at `source/board/mkit_dk_dongle_nrf5x.c` and in particular `nrf_prerun_board_config()`, `sam3u2c_mkit_dk_dongle_nrf5x_if` uses the PB1, PB2 and PB3 pins to detect the board it is running on.

mbrossard, 2023-02-04:
Added in #913

closed 2023-02-04 by mbrossard