ARMmbed/DAPLink #734: RuntimeError: Target m485sidae is not supported.

LagrangeWang, 2020-10-15:
I have followed the steps to Step 4.1. For MDK progen compilation,then when i enter the code "progen generate -t uvision"，it says "RuntimeError: Target m485sidae is not supported. Please add them to https://github.com/project-generator/project_generator_definitions"
then i open the file "projects.yaml" and delete line contain keyword "m48ssidae",i try to "progen generate -t uvision" again and it run normal and says "......as the default Loader is unsafe......"
is there something wrong?

0xc0170, 2020-11-02:
The error comes from progen (definition missing for m48ssidae target). What version of project generator definitions do you have?

To verify, run `pip install -U project-generator-definitions` to fetch the latest. The error should be not there with the latest version.

0xc0170, 2020-11-09:
reopen with an update (if the issue is still relevant).

closed 2020-11-09 by 0xc0170