ARMmbed/DAPLink #736: Microbit DAPlink 0249 BUG?

zfm076, 2020-11-02:
my microbit code:
```
while(1)
{
  microbit.serial.send(microbit.thermometer.getTemperature());
  microbit.sleep(50);    //Send the temperature to the computer every 50ms
}
```
Problem:
In my computer :
It worked all day, The yellow light behind the board also blinks.There is no problem.

But it's on my friend's computer:
Can only work normally for a few minutes or ten minutes,Time is not fixed. The yellow light behind the board is usually on, not flashing,My friend's computer can't get data via USB. Then I change the board back to my computer ,it was able to work a full day. Why? I've been searching for a week and I can't find the reason. Do you have this problem? What do you think is most likely the problem?

Parameter:  
Daplink Mode: Interface
Interface Version: 0249
Bootloader Version: 0243

0xc0170, 2020-11-09:
Does older firmware work (is this specific to 0249?) ?

zfm076, 2020-11-10:
> Does older firmware work (is this specific to 0249?) ?

Interface Version: 0249   ,This is working fine on my computer.

closed 2023-02-04 by mbrossard