ARMmbed/DAPLink #737: DAPLink tests gets hanged on Toshiba Targets

deepak-shreshti, 2020-11-04:
Hi @0xc0170, 
We are facing an issue in AdBun-M066 DAPLink automated testing, not able to perform DAPLink tests after executing the below command:-
`python test\run_test.py --force --notestendpt --testdl --loadbl`

The DAPLink tests (--testdl) gets hanged after progressing to some point.
The Bootloader firmware gets missing from the board after the tests hang, so again and again, Bootloader has to be flashed.
 
**Please suggest, how to fix this issue?**

0xc0170, 2020-11-04:
@flit would you know how to help?

deepak-shreshti, 2020-11-18:
Hi @flit  

Could you kindly check and provide your feedback asap.

Best Regards
Deepak

0xc0170, 2020-11-25:
> The Bootloader firmware gets missing from the board after the tests hang, so again and again, Bootloader has to be flashed.

Just reading the command. if you are executing `--loadbl` I would expect that bootloader is loaded before running the tests. If the tests are already executing, bootloader should be already loaded and checked, so should be valid. Does this mean later tests are corrupting the bootloader ?

How does this behave without `--loadbl` ?

deepak-shreshti, 2020-12-16:
Hi @0xc0170,

The behavior of the tests remains the similar even without `--loadbl`.
DAPLink tests still get hanged expecting the `NEED_BL.TXT` file to be present.
The problem seems to be with `--testdl`, not with `--loadbl`.
Please find the below log without `--loadbl` flag:

[tmpm366_tmpm066_stk_if AdBun-M066.txt](https://github.com/ARMmbed/DAPLink/files/5693364/tmpm366_tmpm066_stk_if.AdBun-M066.txt)

When running tests with `--testdl` as well as `--loadbl`, the tests proceed till some point and then 
remain hanged from then on.

Thanks and Best Regards
Deepak

closed 2023-02-04 by mbrossard