ARMmbed/DAPLink #739: All DAPLink use devices goes away when calling PurgeComm after serial prints any non-ascii characters

pansila, 2020-11-11:
I'm making an automation test script to run a series of commands over serial port provided by DAPLink CDC function.

I noticed the issue happen when the script was calling `PurgeComm`, a kernel32 API on Windows 10, after serial console was sending any non-ascii characters to USB CDC port. If I just print them then everything is fine. The issue happens when I call PurgeComm instead of print any characters in the output buffer.

I guess calling PurgeComm with non-ascii characters in the serial driver output buffer causes DAPLink firmware hang or serial port driver go to a mess.

As a comparsion, I don't see the issue when using a FT232 chip to perform the same operations.

0xc0170, 2020-11-11:
@flit is this anything known to you ?

pansila, 2020-11-12:
After uninstalling the MBED serial driver (mbedWinSerial_16466.exe) to use the Windows 10 provided driver, the issue is gone.

0xc0170, 2020-11-12:
> After uninstalling the MBED serial driver (mbedWinSerial_16466.exe) to use the Windows 10 provided driver, the issue is gone.

Nice find, I recall having other problems when I had it installed previously.

closed 2021-06-08 by mathias-arm