ARMmbed/DAPLink #741: can not use progen generate the project

2502797718, 2020-11-18:
PS F:\DAPLINK\DAPLink> progen generate -f projects.yaml -p stm32f103xb_stm32f746zg_if -t uvision
Traceback (most recent call last):
  File "C:\Python27\Scripts\progen-script.py", line 11, in <module>
    load_entry_point('project-generator==0.9.13', 'console_scripts', 'progen')()
  File "c:\python27\lib\site-packages\project_generator\main.py", line 62, in main
    return args.func(args)
  File "c:\python27\lib\site-packages\project_generator\commands\generate.py", line 30, in run
    if project.generate(args.tool, copied=args.copy, copy=args.copy) == -1:
  File "c:\python27\lib\site-packages\project_generator\project.py", line 552, in generate
    files = exporter(self.project['export'], self.settings).export_project()
  File "c:\python27\lib\site-packages\project_generator\tools\uvision.py", line 490, in export_project
    path, files = self._export_single_project('uvision') #todo: uvision will switch to uv4
  File "c:\python27\lib\site-packages\project_generator\tools\uvision.py", line 452, in _export_single_project
    self._set_target(expanded_dic, uvproj_dic, tool_name)
  File "c:\python27\lib\site-packages\project_generator\tools\uvision.py", line 317, in _set_target
    if not pro_def.is_supported(expanded_dic['target'].lower()):
  File "c:\python27\lib\site-packages\project_generator_definitions\definitions.py", line 129, in is_supported
    mcu_record = self.targets.get_mcu_record(target) if self.mcus.get_mcu_record(target) is None else self.mcus.get_mcu_record(target)
  File "c:\python27\lib\site-packages\project_generator_definitions\definitions.py", line 46, in get_mcu_record
    return _load_record(mcu_path)
  File "c:\python27\lib\site-packages\project_generator_definitions\definitions.py", line 28, in _load_record
    config = yaml.load(project_file, Loader=yaml.FullLoader)
AttributeError: 'module' object has no attribute 'FullLoader'

2502797718, 2020-11-18:
i have solve the problem, but i don not konw the reason.
see the picture, just modify the function's argument.
[image]

moyoumos, 2020-11-23:
FullLoader is a class in PyYAML module and available only in version of PyYAML >=5.1.
try to upgrade PyYAML by:

`pip install -U PyYAML`

0xc0170, 2020-11-23:
Thanks @moyoumos . That is correct.

This is because the version of project generator used here did not define PyYAML version (it was fixed post 0.9.13).

I'll close this as resolved (updating PyYAML fixes the problem).  The best would be if we update project generator to 0.9.17.

closed 2020-11-25 by 0xc0170