ARMmbed/DAPLink #752: Adding a new HIC

Ashleyvk, 2020-12-19:
For adding a new HIC, documentation is not provided in the site. Will editing the default HICs be sufficient?

Ashleyvk, 2021-01-07:
Any suggestions on this?

cederom, 2021-01-07:
Hello @Ashleyvk maybe you will find https://github.com/cederom/DAPLink/tree/cederom_doc a bit more helpful - look at README.md and the doc/ directory of the branch - maybe if you create missing parts they will be accepted into upstream ;-)

Ashleyvk, 2021-01-08:
Hi @cederom in the documentation, the instructions for adding a new board is only provided. Though we can generate the makefile, its compilation requires the source files to be edited. We are planning on using STM32WB55RG & STM32F373CC. Is there any steps to be followed for adding a new HIC?

cederom, 2021-01-08:
Probably you need to take a look at how things are done for other boards and mimic the solution. There is no documentation true but you could be the best person to document that part :-)

Ashleyvk, 2021-01-12:
Hi, 
As per the documentation, when I tried to create the make file, I encountered the following error:

raise RuntimeError("Target %s is not supported. Please add them to https://github.com/project-generator/project_generator_definitions" % expanded_dic['target'].lower())
RuntimeError: Target stm32f373cc is not supported. Please add them to https://github.com/project-generator/project_generator_definitions

For adding a new yaml file in project generator definitions folder, how do we obtain the information required?

closed 2021-01-30 by Ashleyvk