ARMmbed/DAPLink #753: Compiler Issue

Ashleyvk, 2020-12-19:
Im using the Keil free version which has got no interleave option. By default I guess interleave is set as on. Is there a walk around for this? Also can we compile DAPLink in MBED online Compiler or MBED Studio both of which are free ?

cederom, 2020-12-19:
Experimental GCC support is ready for testing: https://github.com/ARMmbed/DAPLink/pull/750

Ashleyvk, 2020-12-21:
I cloned the sources from "https://github.com/mathias-arm/DAPLink.git", provided the GCC path and obtained the following error:
make.py: error: Could not find executable for ARM.
Currently set search path: No path set
It still requires an ARM path. Any suggestion on what can be done?

Ashleyvk, 2020-12-21:
The configuration list was obtained as shown below:

(venv) C:\Users\Ashley\DAPLink_GCC\DAPLink>mbed config --list
[mbed] Global config:
No global configuration is set

[mbed] Local config (C:\Users\Ashley\DAPLink_GCC\DAPLink):
GCC_ARM_PATH=C:\GCC\Files\bin
TOOLCHAIN=GCC_ARM

mathias-arm, 2021-01-27:
The GCC support is in `experimental_compilers` branch. See discussion in #770.

closed 2021-01-27 by mathias-arm