ARMmbed/DAPLink #756: Why there's two pins for swdio for stm32f1

mttbx, 2020-12-23:
I noticed that there's two pins for swdio for stm32f1, on for output and one for input. Why do you do that? Can I just use one pin?

mttbx, 2020-12-23:
solved

closed 2020-12-23 by mttbx

sdyncyang, 2023-01-09:
Hi Mttbx , could you give some infomation about this issue ? i am confused about this issue too, thanks.

mathias-arm, 2023-01-18:
The point of having separate pins is to avoid reconfiguring the GPIO pin from output to input in the middle of the SWD transactions. If you look at the diagrams for SWD [read](https://developer.arm.com/documentation/ihi0031/a/The-Serial-Wire-Debug-Port--SW-DP-/Serial-Wire-Debug-protocol-operation/Successful-read-operation--OK-response-) and [write](https://developer.arm.com/documentation/ihi0031/a/The-Serial-Wire-Debug-Port--SW-DP-/Serial-Wire-Debug-protocol-operation/Successful-write-operation--OK-response-?lang=en) operations, you will see it starts with writing 8 bits and then reading 3 bits.