ARMmbed/DAPLink #757: fail to debug stm32f1 with jtag mode

mttbx, 2020-12-24:
Noticed that I can debug stm32f1 with swd mode.

openocd.cfg:
```
source [find interface/cmsis-dap.cfg]
transport select jtag
source [find target/stm32f1x.cfg]

telnet_port 7777
gdb_port 8888
tcl_port 9999
init
```

result:
```
Debug: 23 3 configuration.c:97 find_file(): found openocd.cfg
Debug: 24 3 configuration.c:97 find_file(): found /usr/local/bin/../share/openocd/scripts/interface/cmsis-dap.cfg
Debug: 25 3 command.c:146 script_debug(): command - adapter driver cmsis-dap
Debug: 27 3 command.c:352 register_command_handler(): registering 'cmsis-dap'...
Debug: 28 3 command.c:352 register_command_handler(): registering 'cmsis-dap'...
Debug: 29 3 command.c:352 register_command_handler(): registering 'cmsis_dap_vid_pid'...
Debug: 30 3 command.c:352 register_command_handler(): registering 'cmsis_dap_serial'...
Debug: 31 3 command.c:146 script_debug(): command - transport select jtag
Debug: 32 3 command.c:352 register_command_handler(): registering 'jtag_flush_queue_sleep'...
Debug: 33 3 command.c:352 register_command_handler(): registering 'jtag_rclk'...
Debug: 34 3 command.c:352 register_command_handler(): registering 'jtag_ntrst_delay'...
Debug: 35 3 command.c:352 register_command_handler(): registering 'jtag_ntrst_assert_width'...
Debug: 36 3 command.c:352 register_command_handler(): registering 'scan_chain'...
Debug: 37 3 command.c:352 register_command_handler(): registering 'runtest'...
Debug: 38 3 command.c:352 register_command_handler(): registering 'irscan'...
Debug: 39 3 command.c:352 register_command_handler(): registering 'verify_ircapture'...
Debug: 40 3 command.c:352 register_command_handler(): registering 'verify_jtag'...
Debug: 41 3 command.c:352 register_command_handler(): registering 'tms_sequence'...
Debug: 42 3 command.c:352 register_command_handler(): registering 'wait_srst_deassert'...
Debug: 43 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 44 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 45 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 46 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 47 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 48 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 49 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 50 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 51 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 52 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 53 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 54 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 55 3 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 56 3 command.c:352 register_command_handler(): registering 'svf'...
Debug: 57 3 command.c:352 register_command_handler(): registering 'xsvf'...
Debug: 58 4 configuration.c:97 find_file(): found /usr/local/bin/../share/openocd/scripts/target/stm32f1x.cfg
Debug: 59 4 configuration.c:97 find_file(): found /usr/local/bin/../share/openocd/scripts/target/swj-dp.tcl
Debug: 60 4 command.c:146 script_debug(): command - transport select
Debug: 61 4 configuration.c:97 find_file(): found /usr/local/bin/../share/openocd/scripts/mem_helper.tcl
Debug: 62 4 command.c:146 script_debug(): command - add_usage_text mrw address
Debug: 64 4 command.c:1123 help_add_command(): added 'mrw' help text
Debug: 65 4 command.c:146 script_debug(): command - add_help_text mrw Returns value of word in memory.
Debug: 67 4 command.c:1136 help_add_command(): added 'mrw' help text
Debug: 68 4 command.c:146 script_debug(): command - add_usage_text mrh address
Debug: 70 4 command.c:1123 help_add_command(): added 'mrh' help text
Debug: 71 4 command.c:146 script_debug(): command - add_help_text mrh Returns value of halfword in memory.
Debug: 73 4 command.c:1136 help_add_command(): added 'mrh' help text
Debug: 74 4 command.c:146 script_debug(): command - add_usage_text mrb address
Debug: 76 4 command.c:1123 help_add_command(): added 'mrb' help text
Debug: 77 4 command.c:146 script_debug(): command - add_help_text mrb Returns value of byte in memory.
Debug: 79 4 command.c:1136 help_add_command(): added 'mrb' help text
Debug: 80 4 command.c:146 script_debug(): command - add_usage_text mmw address setbits clearbits
Debug: 82 4 command.c:1123 help_add_command(): added 'mmw' help text
Debug: 83 4 command.c:146 script_debug(): command - add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 85 4 command.c:1136 help_add_command(): added 'mmw' help text
[... 261 lines cut ...]
Error: 351 419 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 352 425 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 353 431 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 354 437 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 355 443 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 356 449 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 357 455 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 358 461 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 359 467 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 360 473 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 361 479 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 362 485 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Error: 363 491 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Debug: 364 495 arm_adi_v5.c:698 dap_dp_init(): DAP: wait CDBGPWRUPACK
Debug: 365 495 arm_adi_v5.h:506 dap_dp_poll_register(): DAP: poll 4, mask 0x20000000, value 0x20000000
Error: 366 503 adi_v5_jtag.c:425 jtagdp_overrun_check(): Invalid ACK (4) in DAP response
Debug: 367 507 command.c:626 run_command(): Command 'dap init' failed with error code -107
User : 368 507 command.c:692 command_run_line(): 
Debug: 369 507 command.c:626 run_command(): Command 'init' failed with error code -4
User : 370 507 command.c:692 command_run_line(): 
Info : 371 507 server.c:311 add_service(): Listening on port 9999 for tcl connections
Info : 372 507 server.c:311 add_service(): Listening on port 7777 for telnet connections
Debug: 373 507 command.c:146 script_debug(): command - init

```

mttbx, 2020-12-24:
pyocd outptus:
```
Enter the number of the debug probe to connect:
> 1
0001813:WARNING:mbed_board:Unsupported board found 29AA
0001813:INFO:board:Target type is cortex_m
0004226:ERROR:__main__:uncaught exception: 
Traceback (most recent call last):
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 251, in read_dp_result_callback
    return result()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 679, in read_reg_cb
    res = transfer.get_result()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 119, in get_result
    self.daplink.flush()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 610, in flush
    self._read_packet()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 765, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 417, in decode_data
    data = self._decode_transfer_block_data(data)
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 384, in _decode_transfer_block_data
    raise DAPAccessIntf.TransferError()
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 95, in init
    self.read_id_code()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 105, in read_id_code
    self.dpidr = self.read_reg(DP_IDCODE)
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 119, in read_reg
    return self.read_dp(addr, now)
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 234, in read_dp
    return read_dp_cb()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 225, in read_dp_cb
    result = result_cb()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 254, in read_dp_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 251, in read_dp_result_callback
    return result()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 679, in read_reg_cb
    res = transfer.get_result()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 119, in get_result
    self.daplink.flush()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 610, in flush
    self._read_packet()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 765, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 417, in decode_data
    data = self._decode_transfer_block_data(data)
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 384, in _decode_transfer_block_data
    raise DAPAccessIntf.TransferError()
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/__main__.py", line 298, in run
    self._commands[self._args.cmd]()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/__main__.py", line 443, in do_gdbserver
    session = ConnectHelper.session_with_chosen_probe(
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/core/helpers.py", line 199, in session_with_chosen_probe
    session.open()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 157, in open
    self._board.init()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/board/board.py", line 52, in init
    self.target.init()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/core/coresight_target.py", line 136, in init
    seq.invoke()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 189, in invoke
    resultSequence = call()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 100, in init
    self.read_id_code()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 105, in read_id_code
    self.dpidr = self.read_reg(DP_IDCODE)
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 119, in read_reg
    return self.read_dp(addr, now)
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 234, in read_dp
    return read_dp_cb()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 225, in read_dp_cb
    result = result_cb()
  File "/home/xx/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 254, in read_dp_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError

```

mttbx, 2020-12-24:
I can debug it with jlink
```
Open On-Chip Debugger 0.10.0+dev-01405-gc56aa667c (2020-11-30-17:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : J-Link V9 compiled Oct 25 2018 11:46:07
Info : Hardware version: 9.40
Info : VTarget = 3.158 V
Info : clock speed 1000 kHz
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x3ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x3)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x16410041 (mfg: 0x020 (STMicroelectronics), part: 0x6410, ver: 0x1)
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 8888
Info : Listening on port 8888 for gdb connections
Info : Listening on port 9999 for tcl connections
Info : Listening on port 7777 for telnet connections
```

mttbx, 2020-12-24:
So what's the issue, can you help me ?

mttbx, 2020-12-24:
solved.

closed 2020-12-24 by mttbx

lilang1746, 2021-07-04:
你好，请问你是怎么解决的

lilang1746, 2021-07-04:
How did you solve it？