ARMmbed/DAPLink #765: Gang-programming support?

raveslave, 2021-01-11:
For larger panels in production, but also useful for in-system-programming >1 mcu via SWD (when you don't trust your boot-loaders), it would be useful if we could get gang-programming support in DAPLink. 

_there is a cool project called http://cortexprog.com that can do this amongst many other things_

40Grit, 2021-01-11:
From a JTAG standpoint this would be daisy chaining? @flit how does daplink currently handle this for a target like psoc6 with two cores?

nerdralph, 2021-01-21:
cortexprog's gang programming is referring to the ability to use multiple debug probes.  I know pyocd will let you choose between multiple CMSIS-DAP probes based on the USB serial descriptor string reported by the probe.   Although many MCUs have some form of a UUID, it's not part of the Cortex-M spec AFAIK, so it would be dependant on the specific MCU that DAPLink is running on.

flit, 2021-01-28:
You might be able to use JTAG daisy chaining for gang programming, but it would be better to have a multiplexer or multiple IOs. Otherwise the operator wouldn't be able to hot swap boards as they complete programming, since the JTAG chain couldn't be broken.

For multicore devices, except some outliers like the LPC43xx and Raspberry Pi Pico, there is a single DP with multiple APs. In other words, a single SWD or JTAG connection (a DP equates to a TAP when using JTAG).