ARMmbed/DAPLink #766: DAPLink failing to enumerate over USB - STM32F103xB

stephendpmurphy, 2021-01-12:
Hello all :wave:, I'm hoping to get some assistance on my custom board and issues related to USB enumeration.

## Preamble
I am attempting to program a fresh build of DAPLink boards based off of the STM32F103CBT6 micro.. I am building the application using the experimental branch for ARM GCC builds currently: https://github.com/ARMmbed/DAPLink/pull/750

I believe this is an actual app side issue and not related to the compiler, but if someone believes it's related to me using the experimental branch references above, then I will gladly add to the discussion over there.

I am still a bit confused and lost as far as how the process goes for loading a "brand new" board without a bootloader and I apologize if I am missing details in my process.

## Compiling & Flashing
Using the basic setup steps provided in the above PR, I am building both the interface and bootloader fw using the following process.

```
$ python tools/progen_compile.py --clean stm32f103xb_if
$ python tools/progen_compile.py --clean stm32f103xb_bl
```

Both of those compiles great without any errors. :partying_face:. I then load each of the binaries created using my JLink.
- stm32f103xb_bl_crc.bin is loaded at 0x08000000
- stm32f103xb_if_crc.bin is loaded at 0x0800c000

## Bootloader issues
Here's where I start to have problems. The bootloader actually appears to enumerate fine as a drive labeled "Maintenance"... However, it appears to be enumerating as "Read-Only", as I cannot place new "interface" firmware into the drive to update the device.. Not sure what that is about. It did enumerate ONE time and I was able to drag-n-drop and program the device.

```
[  677.265640] usb 1-1: new full-speed USB device number 14 using xhci_hcd
[  677.394102] usb 1-1: device descriptor read/64, error -71
[  677.629912] usb 1-1: device descriptor read/64, error -71
[  677.865912] usb 1-1: new full-speed USB device number 15 using xhci_hcd
[  677.993673] usb 1-1: device descriptor read/64, error -71
[  678.229943] usb 1-1: device descriptor read/64, error -71
[  678.338134] usb usb1-port1: attempt power cycle
[  678.989638] usb 1-1: new full-speed USB device number 16 using xhci_hcd
[  679.011330] usb 1-1: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
[  679.011335] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  679.011339] usb 1-1: Product: DAPLink CMSIS-DAP
[  679.011342] usb 1-1: Manufacturer: Arm
[  679.011344] usb 1-1: SerialNumber: 00000000066aff495655836687172651a5a5a5a597969908
[  679.014064] usb-storage 1-1:1.0: USB Mass Storage device detected
[  679.014670] scsi host0: usb-storage 1-1:1.0
[  680.026641] scsi 0:0:0:0: Direct-Access     MBED     VFS              0.1  PQ: 0 ANSI: 2
[  680.027251] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  680.027716] sd 0:0:0:0: [sda] 131200 512-byte logical blocks: (67.2 MB/64.1 MiB)
[  680.027944] sd 0:0:0:0: [sda] Write Protect is off
[  680.027947] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[  680.028120] sd 0:0:0:0: [sda] No Caching mode page found
[  680.028126] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  680.086440]  sda:
[  680.087958] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

## Interface Issues
I have several more issues with the interface firmware. All related to USB enumeration. The output can be seen below. And actually at the time of writing this issue, I can't 100% produce the output I was seeing before. I was previously seeing errors related to the FAT-FS failing to enumerate (I'm assuming that's the "MSC" portion of the interface trying to enumerate). The CDC port appears to enumerate fine and I can connect a terminal (I haven't attempted to send data over it yet, but that is to be tested soon).
```
[  905.884534] usb 1-1: new full-speed USB device number 17 using xhci_hcd
[  906.013070] usb 1-1: device descriptor read/64, error -71
[  906.248506] usb 1-1: device descriptor read/64, error -71
[  906.484748] usb 1-1: new full-speed USB device number 18 using xhci_hcd
[  906.612552] usb 1-1: device descriptor read/64, error -71
[  906.873157] usb 1-1: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
[  906.873162] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  906.873165] usb 1-1: Product: DAPLink CMSIS-DAP
[  906.873168] usb 1-1: Manufacturer: Arm
[  906.873170] usb 1-1: SerialNumber: 00000080066aff495655836687172651a5a5a5a597969908
[  906.878057] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  906.879370] usbhid 1-1:1.2: couldn't find an input interrupt endpoint
```
## Conclusion
With the errors above and the current state of the bootloader, I believe my hardware is fine (Again, this is a NEW custom board that I designed, based on the ref design for the STM32F103CBT6, but please feel free to ask questions and probe in relation to the hardware design). @mathias-arm appears to be having no issues over on his side with the experimental branch for the ARM GCC compiler so I believe this is an app problem, and or a project setup problem.

Hopefully someone can come along and shed some lights on my very vague understanding of this project. Thus far though, I'm very excited to get this project moving along and start contributing to this community.

Cheers :beers:

stephendpmurphy, 2021-01-14:
Closing this. After further examination and a couple realizations on my part. I believe I have a working board. :+1: 

I have not revisited the odd issues I was seeing with the BL enumerating as a "Read Only" device, but my issues on the app side appear to have been resolved.

closed 2021-01-14 by stephendpmurphy

BibyThm, 2021-01-20:
Hi, I have customized the daplink files for STM32F373CC. The project was compiled using Keil. When it was loaded to the board, the USB was not recognized. It shows "USB is not recognizing the device". Any suggestion to the above problem?

quangcaol, 2024-05-13:
Hi, how did you solve the interface issues. Mine got the same problem as you.