ARMmbed/DAPLink #768: USB not recognized

BibyThm, 2021-01-20:
Hi, I have customized the daplink files for STM32F373CC. The project was compiled using Keil. When it was loaded to the board, the USB was not recognized. It shows "USB is not recognizing the device". Any suggestion to the above problem?

mbrossard, 2021-01-27:
As mentioned in [PR #750](https://github.com/ARMmbed/DAPLink/pull/750#issuecomment-767964899) to @Ashleyvk, you seem to be working on the same MCU in two different environments. Except for the startup part, you should be able to help each other.

BibyThm, 2021-01-29:
Any suggestions to the above problem?

Ashleyvk, 2021-01-29:
I'm facing the same issue too. Has it got something to do with the usbd file for the controller, under the stm32 folder?

Ashleyvk, 2021-03-22:
[image]
When the board is connected, it shows that device descriptor failed. The result of USB analyzer is as attached. Any suggestions on this?

symovs, 2021-03-24:
We are using STM32CubeProgrammer to load the DAPLINK binary in to our custom HIC implemented using STM32F373CC. We are loading the GCC compiled bin file (*_if_crc.bin) with the start address set to 0x0800000 in the STM32CubeProgrammer. 

Are we doing this right ? 
Do we need any DAPLINK specific bootloader for STM32 HIC ? 

@mbrossard @flit thoughts are welcome.