ARMmbed/DAPLink #778: flash_algo.py does not work in python 3

thedjnK, 2021-02-11:
Script seems to be entirely broken with python 3:

```
Traceback (most recent call last):
  File "flash_algo.py", line 359, in <module>
    main()
  File "flash_algo.py", line 48, in main
    algo = PackFlashAlgo(data)
  File "flash_algo.py", line 76, in __init__
    self.elf = ElfFileSimple(data)
  File "flash_algo.py", line 314, in __init__
    super(ElfFileSimple, self).__init__(StringIO.StringIO(data))
AttributeError: type object '_io.StringIO' has no attribute 'StringIO'
```

Not that I have much dealings with python but after trying to fix that issue it just unravels a myriad of further issues with the script and does not work.

elfmimi, 2021-02-12:
@thedjnK what platform are you working on? what is the exact version of your python3? what was the command line? did you try setting up virtualenv ?

thedjnK, 2021-02-12:
@elfmimi Python 3.8.1, windows 10, command used:
```
python .\tools\flash_algo.py .\NewDevice.FLM .\c_blob.tmpl test.c
```

Not using virtualenv no, normal python installation. Not sure how virtualenv would fix it however since the errors relate to byte/string usage mismatch and other issues

elfmimi, 2021-02-13:
OK, I got it working. Please try this.
https://github.com/elfmimi/DAPLink/blob/fix_flash_algo/tools/flash_algo.py

thedjnK, 2021-02-15:
That solves it!

closed 2021-02-15 by thedjnK