ARMmbed/DAPLink #783: Transfer timeout in some linux machines

gerargz, 2021-02-18:
When flashing an image via MSD in Linux Mint 19.1, a FAIL.TXT file is generated with the error: "The transfer timed out.".

This issue was first reported on microbit devices (v0255) but can be replicated on a FRDM-K64F as well. So far, I have only been able to replicate this in Linux Mint 19.1, but some users have reported it on other Linux machines. (@microbit-carlos do you have more info on the other environments where this issue has appeared?)

I have narrowed down the issue as follows:
1. No issues in v0251. Can flash applications and no error is reported.
2. In v0252, there is an issue where the applications don't start after flashing (power cycle is needed) but no error is reported. This seems to have been introduced by #557, which interestingly caused the issue in #590.
3. In v0253, the issue in 2. was fixed by #592, but this also seems to have introduced this very specific timeout issue in some linux machines.

Some extra details on the issue:
-Tested in Ubuntu 16.04 and 20.04, and no issue is seen there.
-No issues with Windows or macOS.
-Small .hex files (24kB) don't have this issue, only with big .hex files (622kB as tested).

microbit-carlos, 2022-02-15:
Hi @gerargz,

> In v0252, there is an issue where the applications don't start after flashing (power cycle is needed) but no error is reported.

How did you trace this? Is there an issue logged somewhere? I didn't find something like this linked to the PRs you've listed.

In my latest tests I was able to get this behaviour in an Ubuntu 16.04 VM, during two out of three different sessions I was seeing the time out error quite consistently, but one of the times it didn't really trigger. Was your Linux Mint environment failing more consistently?

gerargz, 2022-02-17:
Hi @microbit-carlos ,

If I recall correctly, in my Linux Mint environment, I was always getting the timeout error when flashing the big .hex file (622kB). I described the issue with v252 as it seemed relevant to the timeout error, but I didn't submit that as a separate issue since it was later fixed in v253.