ARMmbed/DAPLink #786: DAPLink not rebooting after flashing

davidalo, 2021-02-26:
Hi everyone,

I am trying to use DAPLink project for our custom boards. We currently use GCC ARM as our reference compiler, so, as I saw in #197  there are some developments (thanks @flit) in this area. 

I have built successfully DAPLink for a K20 with our board using [feature/gcc](https://github.com/flit/DAPLink/tree/feature/gcc) , and it is working as expected, it starts, it programs the main MCU and so on. But, I am having an issue, when it flashes the board, DAPLink does not restart.

It seems to be non-fatal but it is. If I upload multiple binaries with different names, the older binaries are not deleted and binaries are kept in the memory. 

This can causes 2 problems:
1. Device can run out of memory.
2. When running mbed tests over the board, the binaries are kept and (I think it is due alphabetical selections) it takes incorrect binaries to run the tests, so it fails!.

By exploring the code, I saw that the problem is here: https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/drag-n-drop/vfs_user.c#L309

If I change this `if` to:
```cpp
    if (daplink_is_interface() && config_ram_get_hold_in_bl()) {
        SystemReset();
    }
```

I saw that it is a change added in v0254 version, see [commit](https://github.com/ARMmbed/DAPLink/commit/932ff5e2ba17431c03a43a3ecf5eebdfbcdc0d97). 

Can anyone give me an idea if I need to configure anything?
Thanks in advance.

mathias-arm, 2021-06-08:
Could you try using the `experimental_compilers` branch? I has seen a lot of work since `flit:feature/gcc` and I don't remember seeing this issue when doing my tests on `k20dx` (I use a FRDM-K64F).

davidalo, 2021-06-08:
I didn't know about that branch. I will try and give you feedback asap.
Thanks for your response.

davidalo, 2021-06-25:
Thanks for your response, @mathias-arm . You were right, I have just tried that branch and it is working as expected!

closed 2021-06-25 by davidalo