ARMmbed/DAPLink #792: [GCC] openocd fails to detect stm3232f102xb_if due to USB HID issue

kasunch, 2021-03-18:
I am using a custom designed ``STM32F103CBT6`` based probe (somewhat similar to ``BluePill`` board design) and flashed with ``stm32f103xb_if`` project firmware.
While the probe works with ``pyOCD`` as expected, ``openocd`` fails to detect the probe.
Note that I am using the following part in the ``openocd`` script.

```
source [find interface/cmsis-dap.cfg]
transport select swd
```

Further investigation reveals that ``openocd`` fails to detect the probe while scanning for USB HID devices.
As shown below, ``syslog`` also shows that there is an error related to HID input interrupt endpoint.

```
Mar 17 15:58:22 thepenguin kernel: [298900.618359] usb 1-1.5.4: new full-speed USB device number 84 using xhci_hcd
Mar 17 15:58:22 thepenguin kernel: [298900.698378] usb 1-1.5.4: device descriptor read/64, error -32
Mar 17 15:58:22 thepenguin kernel: [298900.886131] usb 1-1.5.4: device descriptor read/64, error -32
Mar 17 15:58:22 thepenguin kernel: [298901.074334] usb 1-1.5.4: new full-speed USB device number 85 using xhci_hcd
Mar 17 15:58:22 thepenguin kernel: [298901.154408] usb 1-1.5.4: device descriptor read/64, error -32
Mar 17 15:58:23 thepenguin kernel: [298901.342373] usb 1-1.5.4: device descriptor read/64, error -32
Mar 17 15:58:23 thepenguin kernel: [298901.450956] usb 1-1.5-port4: attempt power cycle
Mar 17 15:58:23 thepenguin kernel: [298902.054344] usb 1-1.5.4: new full-speed USB device number 86 using xhci_hcd
Mar 17 15:58:23 thepenguin kernel: [298902.054515] usb 1-1.5.4: Device not responding to setup address.
Mar 17 15:58:24 thepenguin kernel: [298902.285171] usb 1-1.5.4: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
Mar 17 15:58:24 thepenguin kernel: [298902.285179] usb 1-1.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 17 15:58:24 thepenguin kernel: [298902.285183] usb 1-1.5.4: Product: DAPLink CMSIS-DAP
Mar 17 15:58:24 thepenguin kernel: [298902.285187] usb 1-1.5.4: Manufacturer: Arm
Mar 17 15:58:24 thepenguin kernel: [298902.285190] usb 1-1.5.4: SerialNumber: 00000080066cff554857868367073017a5a5a5a597969908
Mar 17 15:58:24 thepenguin kernel: [298902.302122] cdc_acm 1-1.5.4:1.0: ttyACM1: USB ACM device
Mar 17 15:58:24 thepenguin kernel: [298902.303385] usbhid 1-1.5.4:1.2: couldn't find an input interrupt endpoint
Mar 17 15:58:24 thepenguin mtp-probe: checking bus 1, device 86: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.5/1-1.5.4"
Mar 17 15:58:24 thepenguin mtp-probe: bus: 1, device: 86 was not an MTP device
Mar 17 15:58:24 thepenguin snapd[838]: hotplug.go:199: hotplug device add event ignored, enable experimental.hotplug
Mar 17 15:58:24 thepenguin mtp-probe: checking bus 1, device 86: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.5/1-1.5.4"
Mar 17 15:58:24 thepenguin mtp-probe: bus: 1, device: 86 was not an MTP device
Mar 17 15:58:26 thepenguin ModemManager[929]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.5/1-1.5.4': not supported by any plugin
```

After some tinkering, I disabled ``usb-bulk`` module for ``stm3232f102xb_if`` in ``projects.yaml`` and re-flashed.
Then everything starts to work as expected including ``openocd``.
The syslog output is shown below.

```
Mar 18 09:53:32 thepenguin kernel: [357872.086626] usb 1-1.1: new full-speed USB device number 10 using xhci_hcd
Mar 18 09:53:32 thepenguin kernel: [357872.166640] usb 1-1.1: device descriptor read/64, error -32
Mar 18 09:53:32 thepenguin kernel: [357872.354639] usb 1-1.1: device descriptor read/64, error -32
Mar 18 09:53:32 thepenguin kernel: [357872.542652] usb 1-1.1: new full-speed USB device number 11 using xhci_hcd
Mar 18 09:53:32 thepenguin kernel: [357872.622639] usb 1-1.1: device descriptor read/64, error -32
Mar 18 09:53:33 thepenguin kernel: [357872.810652] usb 1-1.1: device descriptor read/64, error -32
Mar 18 09:53:33 thepenguin kernel: [357872.918802] usb 1-1-port1: attempt power cycle
Mar 18 09:53:33 thepenguin kernel: [357873.522918] usb 1-1.1: new full-speed USB device number 12 using xhci_hcd
Mar 18 09:53:33 thepenguin kernel: [357873.523147] usb 1-1.1: Device not responding to setup address.
Mar 18 09:53:34 thepenguin kernel: [357873.755467] usb 1-1.1: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
Mar 18 09:53:34 thepenguin kernel: [357873.755474] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 18 09:53:34 thepenguin kernel: [357873.755478] usb 1-1.1: Product: DAPLink CMSIS-DAP
Mar 18 09:53:34 thepenguin kernel: [357873.755481] usb 1-1.1: Manufacturer: Arm
Mar 18 09:53:34 thepenguin kernel: [357873.755484] usb 1-1.1: SerialNumber: 00000080066cff554857868367073017a5a5a5a597969908
Mar 18 09:53:34 thepenguin kernel: [357873.770613] cdc_acm 1-1.1:1.0: ttyACM1: USB ACM device
Mar 18 09:53:34 thepenguin kernel: [357873.773849] hid-generic 0003:0D28:0204.004B: hiddev2,hidraw4: USB HID v1.00 Device [Arm DAPLink CMSIS-DAP] on usb-0000:00:14.0-1.1/input2
Mar 18 09:53:34 thepenguin mtp-probe: checking bus 1, device 12: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1"
Mar 18 09:53:34 thepenguin mtp-probe: bus: 1, device: 12 was not an MTP device
Mar 18 09:53:34 thepenguin snapd[838]: hotplug.go:199: hotplug device add event ignored, enable experimental.hotplug
Mar 18 09:53:34 thepenguin mtp-probe: checking bus 1, device 12: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1"
Mar 18 09:53:34 thepenguin mtp-probe: bus: 1, device: 12 was not an MTP device
Mar 18 09:53:36 thepenguin ModemManager[929]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1': not supported by any plugin
```

Looks the ``usb-bulk`` module interferes with ``usb-hid`` module.
Any idea about why this happens?

elfmimi, 2021-04-16:
what version of openocd did you use?
openocd-0.11.0 should support bulk mode hid. older versions don't .

kasunch, 2021-04-26:
I am using openocd v0.10.0. 
So, that would be the reason for this issue.
I will check with openocd v0.11.0 and update this issue.

mathias-arm, 2021-05-27:
@kasunch: were you able to confirm v0.11.0 fixes the issue?

kasunch, 2021-05-30:
@mathias-arm : Sorry for being late to reply. Just tested with openocd v0.11.0. Seems to be working fine now.

closed 2021-05-30 by kasunch