ARMmbed/DAPLink #798: [GCC] Build Errors

Ashleyvk, 2021-03-30:
[image]
[image]
[image]

While building the project using gcc, we encountered the following errors with pre and post build script. 
version_git.h file is not generated and has to be added manually

symovs, 2021-04-01:
@flit @mathias-arm Just a note - .bin file generated by GCC compiler is not working. Only the .hex file generated is working. Maybe the .bin file is same as the base file generated for KEIL which should not be used. 

[image]

mbrossard, 2021-06-11:
@Ashleyvk: GitHub Actions work with [Windows](https://github.com/mbrossard/DAPLink/actions/workflows/windows.yml). You can look at the logs and GitHub's [windows-2019](https://github.com/actions/virtual-environments/blob/main/images/win/Windows2019-Readme.md) configuration to see if you can figure out what is wrong.
@symovs: You should use the `*_if_crc.bin` or `*_if_crc.hex` files. The `*_if.bin` file does not have the CRC computed.

mathias-arm, 2021-08-06:
Closing all STM32WBxx issues. Use #835 or start a STM32WBxx discussion.

closed 2021-08-06 by mathias-arm