ARMmbed/DAPLink #810: Little bug in all hic_hal usb_config.c files

asier70, 2021-05-17:
I spent a few days to compile experimental branch of DAPLink for STM32F103 using GCC on eclipse IDE with GNU MCU Eclipse Plug-ins.
I found 2 bugs in every usb_config.c files in hic_hal folder (different file for each platform).
First bug is missed redefinition of WINUSB_INTERFACE so other files don't see this definition properly using GCC.
```
//     Microsoft OS Descriptors 2.0 (WinUSB) support

#ifndef WINUSB_INTERFACE
#define WINUSB_INTERFACE 0
#else
#define WINUSB_INTERFACE 1
#endif
```
Second bug appered when I compiled version without MSC - there was not string description for USB BULK interface.
```
//is:
#define USBD_BULK_IF_STR_NUM       (3+USBD_STRDESC_SER_ENABLE+USBD_ADC_ENABLE*3+USBD_CDC_ACM_ENABLE*2+USBD_HID_ENABLE+USBD_WEBUSB_ENABLE+USBD_BULK_ENABLE)

//should be:
#define USBD_BULK_IF_STR_NUM       (3+USBD_STRDESC_SER_ENABLE+USBD_ADC_ENABLE*3+USBD_CDC_ACM_ENABLE*2+USBD_HID_ENABLE+USBD_WEBUSB_ENABLE+USBD_MSC_ENABLE)

```
Finally I can compile DAPLink v2 without MSC and without bootloader that fits in STM32F103C8T6 :)

mbrossard, 2021-05-19:
I added the two changes you mention to the draft PR #804.

Are you porting to STM32F103C8T6 "Blue Pill"?

asier70, 2021-05-19:
Thanks for reaction.
I have CMSIS-DAP programmer based on STM32F103C8T6 so I wanted to check how fast it works using BULK instead of HID transfer. I disabled bootloader in header file for linker script and disabled usage of MSC (update would not work anyway) and HID (not work when bulk mode is active in this microcontroller). Then firmware fits in 64kB of flash and works :)

mathias-arm, 2021-05-20:
The suggested changes added in #804 were merged into the `experimental_compilers` branch, so you can probably close this issue.

If you share a link to your changes, others might be interested in them.

closed 2021-05-20 by asier70