ARMmbed/DAPLink #818: Windows on ARM, Support?

sidd-kishan, 2021-06-06:
Please let us know when can we have an ARM64 version for Windows on ARM OS. We can help you test We have Windows on Rasberry Pi setup. Please pursue it we at Windows on Rasberry Pi community will be glad to extend support in testing your drivers and tools for ARM64.

mathias-arm, 2021-06-07:
You might want to test compiling DAPLink firmwares (but this more of a test that `arm-none-eabi-gcc` and Python work), or you might want to test that probes with DAPLink firmware work with debuggers running on Windows on Arm.

This question was asked on ~40 projects and looks like spam. I will close this ticket.

closed 2021-06-07 by mathias-arm

sidd-kishan, 2021-06-08:
Dear @mathias-arm ,
                                  Kindly do not consider it as spam, We at Windows on Rasberry Pi community are focused on leveraging the cheap, versatile, and open-source nature to bring Windows on ARM on this platform. 

To enable the cheapest windows machine which in this day and age of pandemic poor countries like india people are not able to afford even the cheap notebook ~$150 range and families who have only television sets at home as an electric appliance would not be able to invest into digital education of their children.

Other SBC x86/x86_64 boards are more expensive than Rasberry Pi and might still be out of reach of an average household hence so many similar questions on multiple git repo if we have enough hardware support for Windows on ARM ARM64 support we can urge Microsoft in an open letter to create a license specifically for such devices.

It will be again a challenge if MS understands it as a challenge to their MS Surface SKU notebooks but that is a challenge for some other day. It just to create awareness in the hardware dev community to improve the Windows on Rasberry Pi platform to enable digital education, research, and prototyping even on a shoestring budget.

mathias-arm, 2021-06-08:
The code from this repository runs on microcontrollers, so there will not be a "ARM64 version for Windows on ARM OS" as there is no version that runs on Linux, MacOS and Windows (on any architecture). It can be compiled on Linux, MacOS and Windows (non-exhaustive list), but then it is installed on small devices that can then be used by applications running Linux, MacOS and Windows. But you would not run DAPLink itself on Windows on Arm.

Let me expand on what I mentioned earlier:
- If you want to compile DAPLink, check that GNU Arm Embedded Toolchain Downloads (we sometimes call it `arm-none-eabi-gcc`) is ported to Windows on Arm. The source can be downloaded [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads). We use the `9-2020-q2-update` version. Building Daplink also requires Python (>3.7) and GNU/Make. You should build DAPLink from the [`experimental_compilers`](https://github.com/ARMmbed/DAPLink/tree/experimental_compilers) branch for the source and instructions.
- If you want test DAPLink with tools running on Windows on Arm, you could test using [pyocd](https://github.com/pyocd/pyOCD) with a DAPLink-enabled development board. There is an [example](https://os.mbed.com/docs/mbed-os/v6.11/debug-test/debug-microbit.html) on how to debug [MbedOS](https://os.mbed.com/) applications running on [BBC micro:bit](https://microbit.org/).

Well wishes on your quest for improved support for Windows on affordable Arm-powered systems.