ARMmbed/DAPLink #824: STM32F103 HIC nRESET pin pullup issue

asier70, 2021-06-11:
I tested DAPLink on HIC with STM32F103CB without problems. Using HIC with clone GD32F103CB causes problem - it starts in bootloader update mode. I investigated this issue and I found the reason.
At startup bootloader checks nRESET pin to determine to go in update mode if this pin is shorted to ground. In theory nRESET pin is configured to output OD state with pullup. I checked datasheet and found that unfortunatelly pins in STM32F103 (and also in clones) have hardware disabled internal pullup or pulldown when they are in in output state. This mode can't be enabled like in other microcontrollers series like STM32F0.
nRESET pin at startup (after configuration) acts as floating pin with undetermined state. Statistically on STM32F103 it is read as Hi state so it seems that programmer works, on GD32F103 it almost always is read as Lo state so programmer goes to update mode.
The best solution is to rewrite code and define nRESET pin output states as follows:
Lo - output OD mode with Lo state
Hi - input mode with pullup
The simplest solution I've found is to change configuration in "hic_hal\stm32\stm32f103xb\gpio.c" file from
```
    // reset button configured as gpio open drain output with a pullup
    HAL_GPIO_WritePin(nRESET_PIN_PORT, nRESET_PIN, GPIO_PIN_SET);
    GPIO_InitStructure.Pin = nRESET_PIN;
    GPIO_InitStructure.Mode = GPIO_MODE_OUTPUT_OD;
    GPIO_InitStructure.Pull = GPIO_PULLUP;
    HAL_GPIO_Init(nRESET_PIN_PORT, &GPIO_InitStructure);
```
to
```
    // reset button configured as gpio input with a pullup (STM32F103 doesn't have pullup in output OD mode)
    HAL_GPIO_WritePin(nRESET_PIN_PORT, nRESET_PIN, GPIO_PIN_SET);
    GPIO_InitStructure.Pin = nRESET_PIN;
    GPIO_InitStructure.Mode = GPIO_MODE_INPUT;
    GPIO_InitStructure.Pull = GPIO_PULLUP;
    HAL_GPIO_Init(nRESET_PIN_PORT, &GPIO_InitStructure);
```
This pin is reconfigured to output OD mode in functions DAP_SETUP() and PORT_SWD_SETUP() so this solution is safe and allows to check nRESET pin real state at startup. For me it works.

homewsn, 2022-01-25:
I had the problem you described with an STM32-MINI board with the original STM32F103CBT6 chip. Your solution worked great. Thank you.

reuter188, 2025-07-04:
I can confirm this solution works with custom hardware also using STM32F103CBT6.
Many thanks.