ARMmbed/DAPLink #826: Programmer and utility program

robschw, 2021-06-20:
Hallo!
Have some programmed a little PC-program (e.g. QT, JAVA, python, ..) to program NXP (e.g. LPC11xx, LPC15xx) ?

davidalo, 2021-06-25:
Could you explain your question better?

Kevincoooool, 2021-06-26:
segger jflash

robschw, 2021-06-27:
Hallo!
Segger is an excellent company when it comes to JATD / SWD programmer. The tools are perfect. However, the devices come at a fair price for businesses. 
DAPlink is an inexpensive and "convenient" alternative to the well-known JATG / SWD programmers. DAPlink also works very well with NXP processors, for example. If you now have to program a 20 prototype for a study project, it would be practical to have a programmer tool with which you can program via e.g. Python, JAVA, etc. via REDlink server, pyOCD or others using a Windows interface.

I think this would be a useful addition to this great project.

Kevincoooool, 2021-06-28:
> Hallo!
> Segger is an excellent company when it comes to JATD / SWD programmer. The tools are perfect. However, the devices come at a fair price for businesses.
> DAPlink is an inexpensive and "convenient" alternative to the well-known JATG / SWD programmers. DAPlink also works very well with NXP processors, for example. If you now have to program a 20 prototype for a study project, it would be practical to have a programmer tool with which you can program via e.g. Python, JAVA, etc. via REDlink server, pyOCD or others using a Windows interface.
> 
> I think this would be a useful addition to this great project.
> 
> Best regards
> Robert

J-flash already support DAPlink now

Hoohaha, 2021-06-28:
Segger JLink already support DAPLink directly since version 7.0.

davidalo, 2021-06-28:
pyOCD supports DAPLink too.