ARMmbed/DAPLink #831: LPC55S69_IF and MCUXpresso

robschw, 2021-07-08:
Hallo!
I look and “play” with DAPlink – develop, because I am interesting into LPC55S69 for DAPlink. (for private study project).

I want to gib some issue.
I hope ist not ebnouch :).

(1)	In your readme.md develop is an issues:
(1a) in “**Step 1.** Initial setup.” does not work “git checkout develop” on Windows10
(1b)  typing mistake: “ ## Activate virtual environment / **Step 2.**”
$ venv/Scripts/activate.bat   (For Windows) ->  venv\Scripts\activate.bat   (For Windows)

(2)	LPC55S69_IF and MCUXpresso
I try your develop in my develop system (Win 10, VM (Win10) for generate project)
I make Eclipe project in MCUXpresso of LPC55S69_BL and _IF.
Both can compile very well. Then the firmware (BL) was programmed to MCU-LINK hardware.
I want to use DAPLink-develop LPC55C69 with MCUXpresso. 
But MCUXpresso will connect the daplink always with JTAG.
I have also tried to disable JTAG in DAP_config.h:
```
 	/// Indicate that JTAG communication mode is available at the Debug Port.
/// This information is returned by the command \ref DAP_Info as part of <b>Capabilities</b>.
#define DAP_JTAG                0 // [RS 2021.07.08]  1               ///< JTAG Mode: 1 = available, 0 = not available.

/// Configure maximum number of JTAG devices on the scan chain connected to the Debug Access Port.
/// This setting impacts the RAM requirements of the Debug Unit. Valid range is 1 .. 255.
#define DAP_JTAG_DEV_CNT        0 // [RS 2021.07.08]  4               ///< Maximum number of JTAG devices on scan chain

/// Default communication mode on the Debug Access Port.
/// Used for the command \ref DAP_Connect when Port Default mode is selected.
#define DAP_DEFAULT_PORT        1               ///< Default JTAG/SWJ Port Mode: 1 = SWD, 2 = JTAG.
```

TERMINAL: 
```
MCUXpresso IDE RedlinkMulti Driver v11.3 (Mar 30 2021 16:50:12 - crt_emu_cm_redlink build 18)
Found chip XML file in C:/Users/XXXX/OneDrive/_Projekte_XXXXXXX/MCUXpresso_demos/WS_LPC1549_SWO_demo/Z_LPCopen_test_SWO_001/Debug\LPC1519.xml
Reconnected to existing LinkServer process.
Connecting to probe 1 core 0 (using server started externally) reports:
'Ee(42). Could not connect to core.'
Retrying...
Reconnected to existing LinkServer process.
Server OK but no connection to probe 1 core 0 (after 3 attempts) - Ee(42). Could not connect to core.
============= SCRIPT: LPC15RunBootRomConnect.scp =============
Error: Hardware interface transfer error
Error: Wire not connected
Error: Wire not connected
Error: Wire not connected
Error: Wire not connected
Halt on reset vector catch
Error: Wire not connected
Error: Wire not connected
Error: Wire not connected
Error: Wire not connected
Error: Wire not connected
Halt on watchpoint write of ROMWSCTRL register
Error: Wire not connected
Error: Wire not connected
============= END SCRIPT =====================================
Failed on connect: Ee(42). Could not connect to core.
No connection to chip's debug port
```

CONSOLE – REDLINK:
```
[Closed]

[Started server]
[Connected on port 3025]
redlink> ProbeList
Index = 1
Manufacturer = Arm
Description = CMSIS-DAP v1
Serial Number = 00000080f73f9e125231fe3851580eaa3016cc084c504355
VID:PID = 0D28:0204
Path = \\?\hid#vid_0d28&pid_0204&mi_02#9&23a6fe2b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
redlink> ProbeStatus
Index = 1
Manufacturer = Arm
Description = CMSIS-DAP v1
Serial Number = 00000080f73f9e125231fe3851580eaa3016cc084c504355
VID:PID = 0D28:0204
Path = \\?\hid#vid_0d28&pid_0204&mi_02#9&23a6fe2b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
IsOpen = FALSE
WireInitialized = FALSE
WireProtocol = JTAG
CoresConfigured = FALSE
PacketSize = 64
Reference Count = 0
HasSWV = FALSE
HasETM = FALSE
HasJTAG = TRUE
HasSWD = TRUE
Probe Type = CMSIS-DAP
Probe Reference Count = 0
redlink> ProbeIsOpen 1
FALSE
redlink> ProbeOpenByIndex 1 
Probe Handle 1 Open
redlink> WireIsConnected 1
FALSE
redlink> WireSwdConnect 1
Error reading from server (redlinkserv.exe):
Read timed out
Error reading from server (redlinkserv.exe):
Read timed out
[Closed]

[Connected on port 3050]
[Closed]

[Closed]

[Connected on port 3075]
[Closed]

[Closed]
```

pictures available

(3)	Asserfile wenn jump to bootloader again wich jumper
If I jump to bootloader again, with jumper on “Short To Disbale VCOM”, a file is apperars:

Content:
```
 	Assert
 	File: ../source/daplink/drag-n-drop/vfs_manager.c
 	Line: 360
 	Source: Application
```

My contribution is probably a very small one, but when I use something, I want to give my gratitude back with a few reminders in order to make a project perfect. I would like to say again a big thank you for this project.

best regards
Robert

robschw, 2021-07-13:
Hallo!
I tried to connect in MCUXpresso with JTAG, it doesn't work.
I user BL and IF from your compilers test status.
best regards
Robert

mathias-arm, 2021-09-07:
I have reproduced the problem.

I have recently found that it might be related to the `probetable.csv` which is used to detect probes. As we are using the same Vendor ID / Product ID as the MCU-Link, the DAPLink firmware might not match the expectation of the current configuration. There is  an older description of the format ([from forum entry](https://community.nxp.com/t5/LPCXpresso-IDE/Auto-detect-custom-CMSIS-DAP-probe/td-p/560832)):

```
VID     PID     Packet Size Has SWD  Has JTAG Has SWV Has ETM Descriptor   Usage Page Interface
------- ------- ----------- -------  -------- ------- ------- ----------   ---------- ---------
0x1FC9, 0x0090, 1024,       1,       1,       0,      0,      "CMSIS-DAP", 0xFF00,    0
 
 
VID:         Vendor ID
PID:         Product ID
Packet Size: USB packet size. Most probes use 64 byte HID packets. The Keil (non-bridge) firmware
             uses a 1024 byte packet. The NXP bridge and non-bridge CMSIS-DAP probes also use a
             1024 byte packet.
Has SWD:     Probe supports SWD protocol.
Has JTAG:    Probe supports JTAG protocol.
Has SWV:     Probe supports Serial Wire Viewer (NXP proprietary)
Has ETM:     Probe supports ETM (NXP proprietary)
Descriptor:  Text string (or partial text string) used to match against the probe product string.
             A non-null descriptor string is used for bridge probes to inform the server it must
             also match the debug connection on Usage Page or Interface number, depending on host.
             If used, both must be valid.
Usage Page:  Usage page, or 0 (zero) if donít care.
Interface:   Interface number, or -1 if donít care. See Descriptor.
```

There has been a recent update which adds an extra field ([from forum entry](https://community.nxp.com/t5/MCUXpresso-IDE/Custom-CMSIS-DAP-probe/m-p/1316057)):
```
The new parameter allows probe filtering based on the USB Device Release Number which is part of the USB device descriptor.

It is mainly used to exclude the non-debug interfaces of MCU-Link probe configurations with no SWD capabilities from being reported as a debug probe.```