ARMmbed/DAPLink #832: config_rom_init Function Error

Ashleyvk, 2021-07-12:
While testing the customized code for stm32wb55rg, the code gets stuck at if (CFG_KEY == config_rom.key) condition in config_rom_init function. Where does the value for config_rom.key gets assigned? @mbrossard @asier70

asier70, 2021-07-13:
Key value in config_rom struct is checked to determine if this struct was programmed in flash memory or not. If not then default struct is programmed into memory. Struct config_rom is placed in flash section:
static volatile cfg_setting_t config_rom __attribute__((section("cfgrom")));
This section is located at address defined in hic_hal/stm32/stm32wbxx/daplink_addr.h
#define DAPLINK_ROM_CONFIG_USER_START   0x080FB400 // Start address of Initialized User Data in Flash Memory
#define DAPLINK_ROM_CONFIG_USER_SIZE    0x00004C00 // Size of Initialized User Data in Flash Memory
Maybe this memory is not available in your microcontroller (not enough flash memory) or is protected or something else. Reading of this memory causes some error/event.

asier70, 2021-07-14:
During debug check address of config_rom struct and compare to DAPLINK_ROM_CONFIG_USER_START address. maybe you have a problem with configuration under your IDE.

Ashleyvk, 2021-07-15:
I've compiled the code using gcc. I've got jlink debugger which can be connected to the controller via JTAG.  Which IDE can be used for debugging?

Ashleyvk, 2021-07-15:
I tried changing the address and size in flash, but did not work.

mathias-arm, 2021-08-06:
Closing all STM32WBxx issues. Use #835 or start a STM32WBxx discussion.

closed 2021-08-06 by mathias-arm