ARMmbed/DAPLink #838: MSD flashing of a hex file without an extension times out the file transfer

microbit-carlos, 2021-08-12:
If a hex file without an extension is flashed the transfer times out and it generates a fail.txt file.

Example hex file without extension to replicate: [microbit-house.zip](https://github.com/ARMmbed/DAPLink/files/6977435/microbit-house.zip)

As it takes about a minute or so to time out you can see the file "still present" on the drive until it remounts, and during this time the programme is already running on the micro:bit as usual:

<p align="center">
[image]
</p>
FAILT.TXT:

```
error: The transfer timed out.
type: transient, user
```
Tested with DAPLink 0254 for micro:bit V1. Details.txt:

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9900360136424e45004f3016000000620000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0254
Git SHA: f499eb6ec4a847a2b78831fe1acc856fd8eb2f28
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0x18740bef
Remount count: 2
URL: https://microbit.org/device/?id=9900&v=0254
```

The data itself is correctly written to flash, so this has not been a problem for micro:bit V1, but because V2 detects the DAPLink error it then flashes the small error programme to scroll the error code on the display and erases the user programme.

That being said, this is very low priority for micro:bit, as there are not many cases where a user would flash a hex file without an extension, and if they do is likely they are using advanced editors and/or developer tools.

Nevertheless I thought it's still worth raising in GitHub to let you know about this issue.

elfmimi, 2021-08-21:
As I understand, this is by design. DAPLink's drag-n-drop feature is not fine tracking sector allocation. so it can not distinguish to which file entry the stream corresponds. instead it's relying on file extension for some extent of certainty.

microbit-carlos, 2021-11-03:
Thanks @elfmimi, I suspected it would be something like that.
@mathias-arm / @flit if this is by design feel free to close this issue 👍