ARMmbed/DAPLink #847: micro:bit: Universal hex flashing on develop

jaustin, 2021-09-09:
Based on testing fd42dbddd134e6e6cc1f017368ba4952c8ea180f

Attempting to flash a universal hex (https://tech.microbit.org/software/spec-universal-hex/) such as the attached crashes DAPLink running on a V2.00 board build with GCC

On armcc MSD flashing is 'globally' broken so can't be sure if this problem is also present there.

[uhex-sectors-upy.hex.zip](https://github.com/ARMmbed/DAPLink/files/7138490/uhex-sectors-upy.hex.zip)

cc @gerargz @mathias-arm @microbit-carlos

gerargz, 2021-09-10:
I debugged this yesterday but haven't found a clear cause so far.

I was also having issues when resetting the NRF (soft reset) with the button which appeared to be caused by failing SWD transactions. I'm suspecting that the SWD clock calculations may not be the same in armcc vs arm gcc, and the resulting SWD speed may be higher than what the NRF supports.

I see that in [SW_DP.c](https://github.com/ARMmbed/DAPLink/blob/d9cff962722883135dd99598759ed1a93e246681/source/daplink/cmsis-dap/SW_DP.c)  the SWD sequence was optimized with -O3 for armcc, but is this still valid in arm gcc? I don't know if -O3 was chosen for some specific reason.

I changed it to -O0 and the reset was more stable now, but I think we need to validate that the microbit specific macros of DELAY_FAST_CYCLES=2U and DAP_DEFAULT_SWJ_CLOCK=8000000 are still resulting in the expected speed.

mathias-arm, 2021-09-20:
@jaustin / @gerargz: It works for me with [fb4b229](https://github.com/ARMmbed/DAPLink/commit/fb4b2292981671c6441d08eec8e51c3a685f4390) could you confirm and close the issue?

closed 2021-09-22 by mathias-arm