ARMmbed/DAPLink #890: the HID interface failed when use stream_open()/stream_write()/stream_close()  in develop branch

wjzhang, 2021-11-04:
Hi: 
   I now merge our old DapLink(It's working fine. version DAPLINK_VERSION=253) code to Develop branch(it support ARM GCC toolchain).
   But when use dapjs to flash the target, it'll random failed at some HID packages. 
  the HID write data send(checked it write done). then read, No response from HID interface. 

   Do anyone test the this develop branch with dapjs (compile by GCC,  last commit is: f5fb4b96ba930fdb529eacef7e22c4ed06e96461)?

wjzhang, 2021-11-04:
Sorry. it caused by main_task() stack overflow.

closed 2021-11-04 by wjzhang

mathias-arm, 2021-11-04:
Hi @wjzhang, can you tell us which HIC you are using and if you are flashing `.bin` or `.hex` files? How much did you increase the stack size?

wjzhang, 2021-11-04:
Hi Mathias:
 the HIC is lpc11u35, flash the .hex file. main task stack size  increased
to 896(it is not minimum value. but it works for me).

Best regards!
Zhangwenjin

mathias-arm, 2021-11-05:
LPC11U35 and `.hex` were my guesses: the HIC does not have a lot of RAM free, `.hex` file handling require a few tens of bytes more than `.bin` and flashing through WebUSB is the use-case with the deepest stack I know of.
With #884, 512 bytes will be freed up, if more stack is needed.