ARMmbed/DAPLink #900: Very slow USB CDC data transfer (< 9600 baud)

ccattuto, 2021-11-21:
Hello,

I'm testing the "daplink-serial" example of dapjs (WebUSB serial, https://armmbed.github.io/dapjs/examples/daplink-serial/web.html). I'm using a NRF52-DK board with the DAPLink firmware"0254_sam3u2c_mkit_dk_dongle_nrf5x_0x5000.bin" (from https://daplink.io/) installed on the SAM MCU. Flashing firmware via WebUSB, drag & drop and everything else works just fine.

The test application running on the nRF52 pushes data to the UART at 115200 baud, with hardware flow control enabled. This has been confirmed to work with different virtual COM ports implementations (i.e., J-Link). 

ISSUE: when hardware flow control is enabled, the WebUSB serial receives data at a very low baud rate (9600 or lower) despite the baud rate being set to 115200. No data loss is observed. On disabling hardware flow control, conversely, data loss is observed and the string "DAPJLink:Overflow" appears in the stream.

QUESTION: it seems that the actual serial speed between the MCU and the host computer via WebUSB is fixed to 9600 or lower. How can this be changed ?

Thanks!

ccattuto, 2021-11-22:
The WebUSB bottleneck is in DAP.js: it is the polling frequency of the serial, which by default is set to 10Hz (100ms delay) limiting the bandwidth to ~640 bytes / s (assuming a maximum payload size of 64 bytes). Increasing the polling frequency proportionally makes available more bandwidth (e.g., by setting a 5ms delay with startSerialRead(serialDelay=5) one gets enough link bandwidth to support serial data transfer at 115200 baud.

closed 2021-11-22 by ccattuto

mathias-arm, 2021-11-22:
Thank you for the update.