ARMmbed/DAPLink #902: 0xE00ABE00, 0x062D780D, 0x24084068, 0xD3000040, 0x1E644058, 0x1C49D1FA, 0x2A001E52, 0x4770D1F2

wowyyy, 2021-11-25:
Hello, I am not quite clear about this section of code, looking forward to your reply.

0xE00ABE00, 0x062D780D, 0x24084068, 0xD3000040, 0x1E644058, 0x1C49D1FA, 0x2A001E52, 0x4770D1F2

elfmimi, 2021-12-21:
You can do:
`python3 -c '__import__("sys").stdout.buffer.write(__import__("struct").pack("8I", 0xE00ABE00, 0x062D780D, 0x24084068, 0xD3000040, 0x1E644058, 0x1C49D1FA, 0x2A001E52, 0x4770D1F2))' > algo.bin`
and then
`arm-none-eabi-objdump -b binary -m arm -M force-thumb -D algo.bin`

elfmimi, 2022-02-22:
This has been led to a resolution in other place.  https://github.com/pyocd/pyOCD/discussions/1252

so the issue can be closed.

mbrossard, 2022-02-22:
Thanks @elfmimi

closed 2022-02-22 by mbrossard