ARMmbed/DAPLink #908: Python dependency resolution conflict with PyYaml

microbit-carlos, 2021-12-08:
Installing the current develop requirements.txt file we get the following error:

```
ERROR: pyocd 0.28.3 has requirement pyyaml<6.0,>=5.1, but you'll have pyyaml 6.0 which is incompatible.
```

To find out if there are other conflicts with PyYaml we can generating a new requirements.txt with pip freeze and then by installing PyYaml at the end we get:
```
The conflict is caused by:
    The user requested PyYAML==6.0
    cmsis-pack-manager 0.4.0 depends on pyyaml<7.0 and >=6.0
    project-generator 0.11.2 depends on pyyaml<7.0 and >=6.0
    project-generator-definitions 0.2.43 depends on pyYAML
    pyocd 0.28.3 depends on pyyaml<6.0 and >=5.1
```

So it looks like we have to either downgrade `cmsis-pack-manager` and `project-generator`, or upgrade `pyOCD`.
For PyOCD PyYaml was upgraded in https://github.com/pyocd/pyOCD/commit/aa4ab29cb36140db6c0a9d164fd5d57e6083be6e released in v0.32.1 (the latest release at the time of writting).

I believe the pyOCD v0.30 introduce significant changes, so it's possible that could break the tests.
I've never run these tests, @mathias-arm can you easily run them with an update PyOCD?

mathias-arm, 2021-12-08:
To run tests you need to have a `daplink-validation` firmware for the target (see [daplink-validation](https://github.com/mbrossard/daplink-validation)).

You can either test a firmware you built locally:
```
$ python ./test/run_test.py --firmware kl27z_microbit_if \
  --project-tool make_gcc_arm \
  --targetdir ~/git/daplink-validation/dist \
   --quick --logdir kl27z_microbit-gcc
```

Or you can test on a build archive (release, CI or beta):
```
$ python ./test/run_test.py --firmware kl27z_microbit_if \
  --firmwaredir ~/Downloads/firmware-0257-6ff1ee54/gcc \
  --targetdir ~/git/daplink-validation/dist \
   --quick --logdir kl27z_microbit-gcc
```

The output should look something like:
```
[1638977425.530578] SubTest: Build test configuration
[1638977425.530870] No board to test firmware kl27z_microbit_if
[1638977425.530962] Test configurations to be run:
[1638977425.530986]     0: APP=kl27z_microbit_if BL=<None> Board=Microbitv2 Target=Microbitv2
[1638977425.530995]
[1638977425.531004] All firmware can be tested
[1638977425.531011]
[1638977425.531025] SubTest: kl27z_microbit_if Microbitv2
[1638977425.531080] Board: Name=Microbitv2 Unique ID=9904360258994e45001290120000005c000000009796990b
[1638977425.531097] Application: Name=kl27z_microbit_if
[1638977425.531105] Bootloader: None
[1638977425.531117] Target: Name=Microbitv2
[1638977425.531133] SubTest: load_interface
[1638977425.531147] SubTest: set_mode
[...]
[1638977569.963903] Data matches
[1638977569.965367] Skipping USB test on this platform

Test: kl27z_microbit_if Microbitv2: Pass

All boards passed
```

microbit-carlos, 2021-12-10:
Thank @mathias-arm!

Btw, I'm unable to run the tests with micro:bit V2.x due to this issue: https://github.com/pyocd/pyOCD/issues/1262

Were you able to run the test when you posted those instructions?

mathias-arm, 2021-12-10:
I think I wrote the instructions before I updated pyOCD.

microbit-carlos, 2021-12-10:
With PyOCD 0.28.3 it doesn't have the micro:bit v2 board info and fails differently:
```
[1639158735.710673] SubTest: HID test
Board ID 9904 is not recognized, using generic cortex_m target.
Generic 'cortex_m' target type is selected by default; is this intentional? You will be able to debug most devices, but not program  flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
Traceback (most recent call last):
  File "./test/run_test.py", line 713, in <module>
    main()
  File "./test/run_test.py", line 691, in main
    tester.run_tests(args.quick)
  File "./test/run_test.py", line 245, in run_tests
    test_endpoints(test_configuration, test_info, quick)
  File "./test/run_test.py", line 95, in test_endpoints
    test_hid(workspace, test_info, quick)
  File "/Users/microbit-carlos/workspace/microbit-foundation/DAPLink-mbef-fork/test/tests/test_hid.py", line 49, in test_hid
    addr = ram_region.start + 4
AttributeError: 'NoneType' object has no attribute 'start'
```

microbit-carlos, 2021-12-10:
Adding this board entry to the venv PyOCD source code to v0.28.3 seems to work.
```python
    "9904": BoardInfo(  "micro:bit v2",         "nrf52840",         "l1_nrf52840-dk.bin",   ),
```

Did you do something similar in your environment?

mathias-arm, 2021-12-10:
> Did you do something similar in your environment?

Yes, but probably with "nrf52833" and the pack installed.

mathias-arm, 2021-12-10:
It does work with pyocd 0.32.1
```
$ pyocd -V
0.32.1
$ python ./test/run_test.py --firmware kl27z_microbit_if   --firmwaredir ~/Downloads/firmware-0257-6ff1ee54/gcc   --targetdir ~/git/daplink-validation/dist --quick --logdir kl27z_microbit-gcc
[1639163981.881541] SubTest: Build test configuration
[1639163981.881804] No board to test firmware kl27z_microbit_if
[1639163981.881883] Test configurations to be run:
[1639163981.881904]     0: APP=kl27z_microbit_if BL=<None> Board=Microbitv2 Target=Microbitv2
[1639163981.881913]
[1639163981.881926] All firmware can be tested
[1639163981.881933]
[1639163981.881949] SubTest: kl27z_microbit_if Microbitv2
[1639163981.881986] Board: Name=Microbitv2 Unique ID=9904360258994e45001290120000005c000000009796990b
[1639163981.882031] Application: Name=kl27z_microbit_if
[...]

----- FLASH NEW BINARY -----
[==================================================] 100%
[1639164004.893924]

[...]
[1639164128.567382] Data matches
[1639164128.570044] Skipping USB test on this platform

Test: kl27z_microbit_if Microbitv2: Pass

All boards passed
```

So maybe it is related to the `0255` firmware, the hex file you are trying to flash or the state of your target.

mathias-arm, 2021-12-10:
```
$ pyocd -V
0.32.1
$ pyocd load -u 9904360258994e45001290120000005c000000009796990b CALL_ME_MICROBIT.hex
0009404:INFO:load_cmd:Loading CALL_ME_MICROBIT.hex
[==================================================] 100%
0032812:INFO:loader:Erased 463872 bytes (114 sectors), programmed 463872 bytes (114 pages), skipped 0 bytes (0 pages) at 19.73 kB/s
```

```
$ cat /Volumes/MICROBIT/DETAILS.TXT
# DAPLink Firmware - see https://daplink.io
Build ID: v0257-beta1 (gcc)
Unique ID: 9904360258994e45001290120000005c000000009796990b
HIC ID: 9796990b
Auto Reset: 1
Automation allowed: 1
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0257
Bootloader Version: 0255
Git SHA: 6ff1ee547b47c2da1a8ba291f1b8475aa874d2f5
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x828c6069
Interface CRC: 0x6804f20e
Remount count: 6
URL: https://microbit.org/device/?id=9904&v=0257
```

mathias-arm, 2021-12-10:
It's `0255`

```
$ pyocd load -u 9904360258994e45001290120000005c000000009796990b CALL_ME_MICROBIT.hex
0004331:INFO:load_cmd:Loading CALL_ME_MICROBIT.hex
[                                                  ]   0%0005300:CRITICAL:__main__:flash erase sector failure (address 0x00000000; result code 0x67)
[...]
pyocd.core.exceptions.FlashEraseFailure: flash erase sector failure (address 0x00000000; result code 0x67)
```

```
$ cat /Volumes/MICROBIT/DETAILS.TXT
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9904360258994e45001290120000005c000000009796990b
HIC ID: 9796990b
Auto Reset: 1
Automation allowed: 1
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0255
Bootloader Version: 0255
Git SHA: 1436bdcc67029fdfc0ff03b73e12045bb6a9f272
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x828c6069
Interface CRC: 0x5b5cc0f5
Remount count: 0
URL: https://microbit.org/device/?id=9904&v=0255
```

PR #909 (2021-12-09): python: update pyOCD (and other python dependencies)
Fixes #908