ARMmbed/DAPLink #915: DAPv1 bundled with DAPv2 has compatibility issues

elfmimi, 2021-12-28:
HID interface of DAPLink may not functioning as expected when bulk interface is also present.
It seems there are more than one pattern of behaviors.
I've analyzed these two out of many.
- firmware-0257-798b9e73/gcc/0257_lpc11u35__0x0000.hex
  HID interface does not have any endpoint, because of endpoint shortage.\
  Host must use SET_REPORT and GET_REPORT to interact with this.\
  Not many software is aware of this kind of configuration.\
  Even Linux won't bind hid driver for this device.

- firmware-0257-798b9e73/gcc/0257_lpc55s69__0x10000.hex
  HID interface has both IN and OUT endpoints.\
  Despite having OUT endpoint it will get stuck if try to use it.\
  Host must use SET_REPORT to workaround the problem.

mbrossard, 2022-01-02:
LPC55S69 HIC fix has been merged. `lpc11u35_if` is the only project using the `lpc11u35` HIC that has bulk interface. There might be 3 simple solutions:
- Remove bulk interface from `lpc11u35_if`.
- Remove HID interface from `lpc11u35_if`.
- Remove bulk interface from `lpc11u35_if` and add `lpc11u35_bulk_if` with bulk (instead of HID) interface.

mbrossard, 2022-01-03:
I tried creating a project with bulk interface but without HID interface and it did not work, so as a first step #920 changes `lpc11u35_if` to be HID-only, like all other `lpc11u35`-based projects.

elfmimi, 2022-01-04:
I don't yet know why, but this must be the cause bulk interface without HID interface is failing.
https://github.com/ARMmbed/DAPLink/blob/8dff8e01027fdb23c49734099a61ee1077af0797/source/hic_hal/nxp/lpc11u35/DAP_config.h#L97-L101
Try changing it to 64 , then it should work fine.

mbrossard, 2022-01-04:
Looking at the USB configuration it seems all interefaces are limited to 64 bytes:
```$ grep WMAXPACKETSIZE source/hic_hal/nxp/lpc11u35/usb_config.c | grep -v _HS_
#define USBD_HID_WMAXPACKETSIZE     64
#define USBD_MSC_WMAXPACKETSIZE     64
#define USBD_ADC_WMAXPACKETSIZE     64
#define USBD_CDC_ACM_WMAXPACKETSIZE     16
#define USBD_CDC_ACM_WMAXPACKETSIZE1    64
#define USBD_BULK_WMAXPACKETSIZE     64
[...]```

I took your advice in https://github.com/mbrossard/DAPLink/tree/pr/lpc11u35_if_bulk and it seems to work.

closed 2022-01-05 by mbrossard

eleenest, 2022-01-08:
Just wondering if this issue is only for the lpc11u35_if (maybe a smaller number of hw endpoints available on this MCU?), or if other HICs with a bulk interface are also affected 

(looks like a total of 12 interfaces with - records/usb/usb-bulk.yaml in projects.yaml)?  

Eric

mbrossard, 2022-01-08:
> Just wondering if this issue is only for the lpc11u35_if (maybe a smaller number of hw endpoints available on this MCU?), or if other HICs with a bulk interface are also affected
> 
> (looks like a total of 12 interfaces with - records/usb/usb-bulk.yaml in projects.yaml)?

I think so. The LPC11U35 "[s]upports 10 physical (5 logical) endpoints including one control endpoint" (https://www.nxp.com/docs/en/data-sheet/LPC11U3X.pdf). We should probably document the number of end-points each HIC has.

mbrossard, 2022-01-08:
I looked up [df8e414], it seems the only other that might be problematic are the MAX3262x.

commit 10103f14 (2022-01-05): Remove HID interface from lpc11u35_if (#922, fixes #915)
commit 8f31e4fb (2022-01-05): lpc11u35_if: remove hid from lpc11u35_if (fixes #915)