ARMmbed/DAPLink #929: Delayed DAP command puts DAPLink USB in bad state

gaborcsapo, 2022-01-12:
Hey, we noticed an issue with the usbd_bulk.c file, when a DAP command arrives late on DAPLink.

Here's our situation:
- OpenOCD sends an OUT packet with a regular DAP command. OpenOCD thinks the packet is sent fine.
- The packet is delayed by the Host OS, because the host is doing something funky to reconnect with another unresponsive USB test device. 
- The packet shows up on the USB analyzer only after 5 seconds on the way to DAPLink.
- DAPLink finally receives and processes the OUT packet then writes the DAP response to its HW TX fifo. Normally, OpenOCD would send an IN packet, the response is transferred, and DAPLink gets ready for the next OUT packet on the TX complete interrupt by setting USB_ResponseIdle to 1.
- However, OpenOCD in the meantime thought the connection has an error, it probably has a timeout of <5sec. Instead of sending the IN packet, it will send another OUT packet with the disconnect DAP command.
- DAPLink receives the new OUT packet, executes the DAP disconnect, but usbd_bulk.c doesn't write the response to the HW TX fifo, because USB_ResponseIdle is still zero and the previous response is still in the usb HW fifo anyways.
- OpenOCD sends an IN packet, but it will read the stale response from the first OUT packet, so it's even more confused. OpenOCD will try a few different commands after this, but it will keep getting a stale response, so the connection never stabilizes.

Our workaround is to clear the TX FIFO and reinit the usbd_bulk whenever we receive a disconnect DAP cmd.

```
void USBD_BULK_EP_BULKOUT_Event(U32 event)
{
    U16 bytes_rece;
    uint8_t * rbuf;

    bytes_rece      = USBD_ReadEP(usbd_bulk_ep_bulkout, ptrDataIn, USBD_Bulk_BulkBufSize - DataInReceLen);
    ptrDataIn      += bytes_rece;
    DataInReceLen  += bytes_rece;

    if ((DataInReceLen >= USBD_Bulk_BulkBufSize) ||
        (bytes_rece    <  usbd_bulk_maxpacketsize[USBD_HighSpeed])) {

       #####################
        if (USBD_Bulk_BulkOutBuf[0] == ID_DAP_Disconnect) {
            usbd_bulk_init();
            USBD_ClearEPBuf(USB_ENDPOINT_IN(usbd_bulk_ep_bulkin));
        }
       ######################

        if (DAP_queue_execute_buf(&DAP_Cmd_queue, USBD_Bulk_BulkOutBuf, DataInReceLen, &rbuf)) {
            //Trigger the BULKIn for the reply
            if (USB_ResponseIdle) {
                USBD_BULK_EP_BULKIN_Event(0);
                USB_ResponseIdle = 0;
            }
        }
        //revert the input pointers
        DataInReceLen = 0;
        ptrDataIn     = USBD_Bulk_BulkOutBuf;
```

What do you think of the issue and do you have a better fix? Seems like this file was designed so that it can receive more than one DAP commands before the results are read. Is that actually the intention? If we never expect two DAP commands, we could clear the buffer before servicing the second, right?

flit, 2022-01-12:
That's quite the mess! 😄 I've never seen a packet delayed that long by the host stack.

Honestly, my initial thought is that the problem is in OpenOCD. It should always perform an IN transfer if it has successfully sent an OUT with a DAP command.

Why does OpenOCD time out? If it things the OUT packet was sent successfully, what is timing out? Is it the completion callback for the OUT packet that times out? Meaning OpenOCD really sees the OUT has failed.

I'm not sure how I feel about clearing the TX FIFO on disconnect command.

> Seems like this file was designed so that it can receive more than one DAP commands before the results are read.

That's correct. The original implementation had USB and DAP command processing running on different threads. The CMSIS-DAP spec allows multiple outstanding commands (the maximum is available as the maximum packet count value from [DAP_Info](https://arm-software.github.io/CMSIS_5/develop/DAP/html/group__DAP__Info.html)). This was restructured a while ago in DAPLink to serialise everything through the main thread, both to solve a number of concurrency related issues and reduce the memory usage from having multiple thread stacks. I'd like to return to the multi-threaded design (for those devices that have the memory) since not supporting multiple outstanding commands negatively affects DAP performance.

flit, 2022-01-12:
I'm curious if the same problem happens with pyOCD. It also uses libusb, but through pyusb, which only exposes a synchronous API.

gaborcsapo, 2022-01-12:
> Why does OpenOCD time out?  If it things the OUT packet was sent successfully, what is timing out? Is it the completion callback for the OUT packet that times out? Meaning OpenOCD really sees the OUT has failed.

OpenOCD thinks the OUT packet is sent, but the host OS doesn't actually send it for 5 seconds, because it's busy with some other funky stuff (I'd actually say the real bug is in linux, but we can't fix that), so OpenOCD probably times out waiting for the OUT packet ACK. And if the OUT command fails, it will try to disconnect and connect again.

flit, 2022-01-12:
Can the timeout in OpenOCD just be increased? This would also solve this problem for non-DAPLink-based CMSIS-DAP probes.

To be clear, I'm not opposed to having workarounds for OpenOCD in DAPLink. (There's already a recent one to work around OpenOCD 0.10.x failing to handle the extended DAP_Info response from CMSIS-DAP v2.1.) Just want to make sure we target the right workaround and don't introduce complications.

A potentially safer variant of your workaround would be to use DAP_TransferAbort instead of DAP_Disconnect. This would obviously require changes on the OpenOCD side as well.

> I'd actually say the real bug is in linux

Indeed…

gaborcsapo, 2022-01-13:
I'll check what are the options for increasing the openOCD timeout. I'm not that familiar with the openOCD source so I might need a couple days.

flit, 2022-01-14:
No problem! If you need help, let me know and I'll take a look too—I have some familiarity with that part of OpenOCD.

gaborcsapo, 2022-03-18:
Hey, sorry for the delay. I had some time to look into a nicer solution. Increasing USB_TIMEOUT to 6s in cmsis_dap.c in nlopenocd is also a solution. I'll try to suggest that change, but I'm skeptical if they would take it since it's just a workaround for a linux usb stack issue (which I don't wanna dig into). What do you think?

However, I still think a recovery mechanism would be nice on DAPLink. When openOCD sees a transaction fail dap.do_reconnect is set to true and on the next transaction it will try to connect again. The reconnection takes us to cmsis_dap_swd_switch_seq, which sends the DISCONNECT cmd first and then CONNECT. What do you think about adding a DAP_TransferAbort, which flushes the DAPLink TX FIFO and resets the state of usbd_bulk.c?

gaborcsapo, 2022-03-18:
You can vouch for my openOCD change [here](https://sourceforge.net/p/openocd/tickets/343/)

flit, 2022-03-20:
Let me talk with the CMSIS team to get their view on flushing the command queue on a DAP_TransferAbort vs DAP_Disconnect. They will probably want to be aware of this issue with Linux, too.

gaborcsapo, 2022-04-01:
Seems like OpenOCD will take the changes, so I'll close this issue. Thanks for the help!

closed 2022-04-01 by gaborcsapo