ARMmbed/DAPLink #934: program a DAPlink with a DAPlink

nikiwaibel, 2022-01-24:
is it possible to program a DAPlink with another DAPlink?

i've just seen in the code that _bl and _if images are detected and flashed to the device itself.

i wonder if thee is a way to use a DAPlink to flash the _bl and _if image to another DAPlink.

elfmimi, 2022-02-02:
Technically yes. Practically, it is mostly no.
It is possible only when the operating DAPLink and the target DAPLink are equipped with certain combination of MCUs.

stm32f103xb programming stm32f103xb is one example that is possible.

nikiwaibel, 2022-02-03:
really? i've tried this with stm32f103xb_bl and stm32f103xb_stm32f103rb_if (develop branch builds) without success, when using the drag&drop-usb interface. it tried to update itself, but not the connected other daplink hardware.

the only way it worked (to program the connected other daplink) was to use the daplink-usb interface with openocd and
```
source [find interface/cmsis-dap.cfg]
transport select swd
source [find target/stm32f1x.cfg]
reset_config srst_only
```
custom script …

elfmimi, 2022-02-16:
Right! Now that you mentioned it, I know I told you a lie.
Unless some modification are introduced, the operating DAPLink and the target DAPLink must be from different family. (could be still inaccurate here.)