ARMmbed/DAPLink #939: Unable to erase flash with erase.act on micro:bit v1

atoncetti, 2022-02-06:
I am unable to erase the flash on a micro:bit v1 by using `erase.act` MSD command after accessing MAINTENANCE mode.

The device reboots into normal MICROBIT mode and the file ASSERT.TXT appears, contents: 
```
Assert
File: (file)
Line: 108
Source: Bootloader
```

DETAILS.txt file:

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0000000052374e45005e100b000000160000000097969901
HIC ID: 97969901
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Bootloader
Bootloader Version: 0243
Interface Version: 0249
Git SHA: b403a07e3696cee1e116d44cbdd64446e056ce38
Local Mods: 0
USB Interfaces: MSD
Bootloader CRC: 0x32eb3cfd
Interface CRC: 0xcdb7b2a3
Remount count: 0
```

Steps to reproduce:

- Plug in the micro:bit v1 while pressing the reset button to enter in MAINTENANCE mode
- Mount MAINTENANCE
- `touch erase.act` inside the mounted dir

carlosperate, 2022-02-09:
Hi @atoncetti,

I believe you need to use the `erase.act` file in the `MICROBIT` drive, not the `MAINTENCE` drive.

If you don't have automation enabled in `MICROBIT` mode, you can simply copy the file **while** the reset button is pressed down. This should then result in a the `MICROBIT` drive remounting and the target flash contents erased.

atoncetti, 2022-02-09:
Hi @carlosperate,

Thanks for your help!
Copying the file while pressing the reset button in `MICROBIT` mode worked.

Cheers!

closed 2022-02-09 by atoncetti