ARMmbed/DAPLink #941: [Renesas GR-PEACH and GR-LYCHEE] USB drag & drop failed by timeout error

toyowata, 2022-02-15:
USB drag & drop failed by timeout error on v0254 or later version (v0253 didn't have this error).

## Target
* Renesas GR-PEACH
* Renesas GR-LYCHEE

## DAPLink version
* v0254 or leter

## How to reproduce
* Flash DAPLink v0254 or later version
* USB drag & drop program e.g. mbed-os-example-blinky
* The mass storage drive is unmounted/mounted (but takes long time)
* FAIL.TXT file is created in the drive and the program is not flashed

FAIL.TXT
```
error: The transfer timed out.
type: transient, user
```

elfmimi, 2022-02-16:
Hi, snowman-san!

What about latest v0257-rc1 ? > [v0257-rc1 [c782a5ba] (2022-02-04)](https://github.com/ARMmbed/DAPLink/wiki/compilers-test-status)
Well I'm guessing "later version" is meant to be taken as including main and develop branch.

What OS is being used to do the file copy ?
and don't you have any speculation on the cause of the issue?

toyowata, 2022-02-16:
@elfmimi 

Thanks for your comment.  I have just used the v0257-rc1 (armcc build binary) and the timeout problem still exist.

>Well I'm guessing "later version" is meant to be taken as including main and develop branch.

Yes, correct.

>What OS is being used to do the file copy ?

I used macOS Monterey 12.2.1, Windows 10, and Windows 11.

elfmimi, 2022-02-16:
I'm glad you solved it quick. 😃

closed 2022-02-18 by mbrossard