ARMmbed/DAPLink #949: Could SECTOR_BUFFER_SIZE be changed to 32

yangtechen, 2022-03-01:
https://github.com/ARMmbed/DAPLink/blob/9d5d045879eea27df34f195ef9f5d240ffdb2417/source/daplink/settings/settings_rom.c#L32
https://github.com/ARMmbed/DAPLink/blob/9d5d045879eea27df34f195ef9f5d240ffdb2417/source/daplink/settings/settings_rom.c#L122

Some MCUs like stm32h743 can only program 32 bytes at a time, better change SECTOR_BUFFER_SIZE to 32.

yangtechen, 2022-03-16:
Hey, what do you think about this idea?

flit, 2022-03-20:
Since 32 is smaller than the sector size of all HICs, the simplest change would be just setting `SECTOR_BUFFER_SIZE` to 32. However, because the programming buffer is allocated on the stack, there is some non-zero risk that certain highly RAM-limited HICs (particularly LPC11U35) would overflow the stack.

The lowest risk change would be to wrap `#define SECTOR_BUFFER_SIZE` with an `#if !defined(SECTOR_BUFFER_SIZE)` Any HIC that requires a larger minimum program size can then define `SECTOR_BUFFER_SIZE` in its yaml record.

mbrossard, 2023-01-28:
It seems `DAPLINK_MIN_WRITE_SIZE` could (should?) have been used, but the value for most HIC does not reflect the original intent.