ARMmbed/DAPLink #967: Not able to Drag-and-Drop with BIN Files

SamECU, 2022-06-27:
Hello,

I'm using the LPC55S69 (MCU-Link Hardware) as HIC which I've attempted to port to the MIMXRT1010-EVB as Target.

With the hardware and software and I can debug the target in IDE (MCUXpresso) and the drag-and-drop functionality works with HEX files. 

When I try and program using BIN files I receive "FAIL.TXT : Transfer timeout".

My first thought is that it doesn't recognise the BIN files.

I'm not sure what I need to change in the code to resolve the issue.

Thought / Issues
1. When I generate c_blob.c for flash_blob.c from "MIMXRT1011_QuadSPI_4KB_SEC.FLM" it generates the data below for "Verify":
[image]

When compiling, "0x120000003, // Verify" isn't a valid value.  Should this be deleted or set to "0x00000000, // Verify"?

2. When I generate c_blob.c for flash_blob.c from "MIMXRT1011_QuadSPI_4KB_SEC.FLM" it generates the data below for "sectors_info":
[image]

When compiling there are errors as "sector_info_t sectors_info" is expected.  Should it be changed to as below:
[image]

3. When populating "Target.c" I can find the information supplied with MIMXRT1011_QuadSPI_4KB_SEC.FLM as below:
[image]

Have I populated "Target.c" correctly or is this issue here:
[image]

(I ask as there's 3x RAM-Start addresses.)

Thanks for any help, I really appreciate it.
[Data.zip](https://github.com/ARMmbed/DAPLink/files/8995210/Data.zip)

mathias-arm, 2022-06-28:
> 1. When I generate c_blob.c for flash_blob.c from "MIMXRT1011_QuadSPI_4KB_SEC.FLM" it generates the data below for "Verify":
> When compiling, "0x120000003, // Verify" isn't a valid value. Should this be deleted or set to "0x00000000, // Verify"?

Yes, that is correct.

> 2. When I generate c_blob.c for flash_blob.c from "MIMXRT1011_QuadSPI_4KB_SEC.FLM" it generates the data below for "sectors_info":
> When compiling there are errors as "sector_info_t sectors_info" is expected. 

`tools/generate_flash_algo.py` in `main` branch should generate `sectors_info` with the `sector_info_t` type.

> 3. When populating "Target.c" I can find the information supplied with MIMXRT1011_QuadSPI_4KB_SEC.FLM as below:
> [...]
> (I ask as there's 3x RAM-Start addresses.)

The flash algo is using 0x20000000 adddress and other i.MXRT targets as well.

Have a look at `validate_bin_nvic` in `source/family/freescale/target_reset_mimxrt.c`. If the  MIMXRT1011 also expects a "Flash Configuration Block" at the start of the `.bin` image, you should define `DAPLINK_MIMXRT_TARGET` macro for your board. Otherwise you should look at the initial stack pointer (first 4 bytes in `.bin` file) and check that it properly fits in the RAM region defined.

SamECU, 2022-06-28:
Hello Mattias,
Thanks a lot, I'll look into each of the comments.

Cheers,
Sam

SamECU, 2022-07-03:
Hallo Mathias

1. RAM Regions
I extended the definition of the RAM regions to reflect my memory map:
[image]
[image]
Works without issue for HEX Files that are with this memory configuration.  Changes to the memory map require an update to the target_device.  I'm not sure why the other iMXRT only need to define one address (or perhaps they'll run into similar issues as myself --> change in memory map must be reflected in detail).

2. BIN Files
My BIN files were starting with an offset (generated from an SDK and MBDT).
[image]
After making the changes below I am successfully able to flash with BIN files.
[image]

Thanks a lot for the help and sorry for misspelling your name.

Regards,
Sam