ARMmbed/DAPLink #974: keil mdk doesn't recognize daplink v0257 downloader. (keil mdk 无法识别 daplink v0257 下载器。)

stevezh1998, 2022-07-27:
你好，我在keil mdk V5.37.0.0中使用daplink v0257发现无法识别到下载器，然后我回退到daplink v0254发现是可以正常使用的；所有我在想是不是下载器硬件出了问题导致新版有问题，所有我又用daplink v0254版本的固件在 openocd 中使用，发现也是可以正常使用；因此我现在迷茫了，不知道问题出在了哪里。
[image]

ShaojieHe, 2022-08-14:
I think you can reference this link [Keil Forum](https://community.arm.com/support-forums/f/keil-forum/52868/keil-cannot-find-daplink-with-new-version-firmware)

from this link showed this issue is caused by Keil and its is already scheduled for a fix

我觉得你可以看一下这个链接 上面说了是keil的问题，需要等下次修复

stevezh1998, 2022-08-14:
Thanks for your answer.
谢谢你解答。

ShaojieHe, 2022-08-14:
> Thanks for your answer. 谢谢你解答。

这个链接里面的暂时解决方法是可以用的，这边刚才测试了一下

stevezh1998, 2022-08-14:
刚才试了一下确实可以正常使用了，非常感谢。
[image]

ShaojieHe, 2022-08-14:
很高兴能帮上忙 :) 没问题可以关issue了
happy to help, you can close this issue if no more further question

closed 2022-08-14 by stevezh1998