ARMmbed/DAPLink #975: what does swj_sequence 12 e0 ff 92 f3 09 62 95 2d 85 86 e9 af dd e3 a2 means?

xidianzhuxi, 2022-08-03:
I am using daplink on my fpga which runs a cortex m33 simulation. I noticed that  daplink  will do this thing:
step 1:  send 7 bytes 0xff, send 0x9e 0xe7, send 7 bytes 0xff, send 2 bytes 0x00. then read target idcode.
step 2:  send sequence 0xff 0x75 0x77 0x77 0x67
step 3:  send sequence 0xff 0x92 0xf3 0x09 0x62 0x95 0x2d 0x85 0x86 0xe9 0xaf 0xdd 0xe3 0xa2 etc.
step 4:  read target idcode
I can find step 1 is introduced in the _arm debug interface v5_ , and I can also find 0x9e0xe7 is introduced in arm website which means turn jtag to swd mode.
My Question Is, about sequence send in step 2 and step 3. what does this mean? and where can I lean this?

elfmimi, 2022-08-03:
1. GitHub is not StackOverflow.
2. You should make clear what version and which variant of DAPLink firmware you are using.
3. DAPLink won't send those commands on its own. So it must be the software on the host side that is requesting it.
4. That naturally leads to a question, what was the command you executed?

See *ARM Debug Interface v5.2 Architecture Specification* .
> B5.3.2 Switching from JTAG to dormant state
> 0x33BBBBBA transmitted LSB first.

> B5.3.4 Leaving dormant state
> 0x19BC0EA2 E3DDAFE9 86852D95 6209F392 transmitted LSB first.

Don't forget to close this issue, when you are satisfied with the answer.

closed 2022-08-04 by xidianzhuxi

xidianzhuxi, 2022-08-04:
Thx, you inspired me this is not daplink's work  , I really got this in pyocd source code.