ARMmbed/DAPLink #977: The STM32F103xb_if projects Tag v0257 and after could not be recognized in MDK

xiaohanhan1982, 2022-09-05:
The STM32F103xb_if projects Tag v0257 and after could not be recognized in MDK, the Tag v0254 is OK. Anybody has solutions？

mbrossard, 2022-10-01:
@zjli-2019 [reported](https://github.com/ARMmbed/DAPLink/pull/805#issuecomment-1041151569) that changing `CMSIS_DAP_PRODUCT_NAME` to `"CMSIS-DAP"` solved the issue. Could you test that and report?

penjun71, 2022-10-13:
i use lpc11u35,same problem.
i edit this ,fix this problem!
dap_strings.h
#define CMSIS_DAP_PRODUCT_NAME "DAPLink CMSIS-DAP"

sdyncyang, 2023-01-02:
i met same issue , STM32F103xb_if projects seems worked, due to i can find the CMSIS-DAP v2 and WebUSB: CMSIS-DAP in compture hardware list , but can not find in MDK , i changed CMSIS_DAP_PRODUCT_NAME to "CMSIS-DAP" ,but still not work ...

PR #981 (2023-01-29): Penjun71 patch daplink for mdk
fix lpc11u35 not found in mdk ,fw >0254

Fixes #977. Some tools (including some versions of MDK) require "CMSIS-DAP" to be present in product name. This has been an issue since DAPLink added support for bulk interface (DAPv2) in v0254.
closed 2023-01-29 by mbrossard