ARMmbed/DAPLink #979: The 0257_lpc11u35__0x0000 Tag v0257 and after could not be recognized in MDK

penjun71, 2022-09-16:
base on lpc11u35 DAPLINK ,it can be found on windows 7 device manager ，but keil mdk could not found it .

IDE-Version:
μVision V5.37.0.0
Copyright (C) 2022 ARM Ltd and ARM Germany GmbH. All rights reserved.

Tool Version Numbers:
Toolchain:        MDK-ARM Plus  Version: 5.37.0.0
Toolchain Path:    C:\Keil_v5\ARM\ARMCC\Bin
C Compiler:         Armcc.exe        V5.06 update 7 (build 960)
Assembler:          Armasm.exe        V5.06 update 7 (build 960)
Linker/Locator:     ArmLink.exe        V5.06 update 7 (build 960)
Library Manager:    ArmAr.exe        V5.06 update 7 (build 960)
Hex Converter:      FromElf.exe        V5.06 update 7 (build 960)
CPU DLL:               SARMCM3.DLL          V5.37.0.0
Dialog DLL:         TCM.DLL              V1.56.1.0
Target DLL:             Segger\JL2CM3.dll      V2.99.42.0
Dialog DLL:         TCM.DLL              V1.56.1.0

mbrossard, 2022-10-01:
@zjli-2019 [reported](https://github.com/ARMmbed/DAPLink/pull/805#issuecomment-1041151569) that changing `CMSIS_DAP_PRODUCT_NAME` to `"CMSIS-DAP"` solved the issue. Could you test that and report?

penjun71, 2022-10-10:
i tested it, this cannot solve this problem

penjun71, 2022-10-13:
i edit this ,fix this problem!
dap_strings.h
 #define CMSIS_DAP_PRODUCT_NAME "DAPLink CMSIS-DAP"

closed 2022-10-13 by penjun71