ARMmbed/DAPLink #982: macOS ventural finder drag&drop flashing fails

microbit-carlos, 2022-10-26:
More info coming.

jaustin, 2022-10-26:
Can confirm on an M1 mac running macOS Version 13.0 (22A380)
In finder I immediately get :

"The operation can’t be completed because an unexpected error occurred (error code 100093)."

I then get a long delay (during which the old program keeps running), then after some time, the device unmounts, returns with a FAIL.txt and micro:bit is showing a 504 error.

Contents of Fail.txt
```
error: The transfer timed out.
type: transient, user
```

(for any micro:bit users finding this - web USB still works in Chrome from the MakeCode or Python editors)

microbit-carlos, 2022-10-27:
DAPLink MSD drive is a "fake" volume, which ignores most of the file system operations other than file writes, as it can then flash a target microcontroller if a .hex or .bin file are copied into the drive.

This is a common mode of operation for tools in this category. I've tried the following and the issue is also present on these products/tools/platforms:

- RP2040 ROM bootloader: https://github.com/raspberrypi/pico-sdk/issues/1081
- J-Link running on a Nordic nRF52840-DK board
- J-Link firmware running on a micro:bit
- OpenSDA running on a FRDM-KL27Z board

microbit-carlos, 2022-10-27:
Some additional data captured by @martinwork:

> My tests were with V2.21 or V1.3B attached to a USB A port, I get 504 for any file dragged to the MICROBIT drive, and the mac log contains
> 
> ```
> UVFSService LFSAD: loadADHeader: magic is wrong: 0, 0, 0
> DesktopServicesHelper TCFURLInfo:TranslateRawPOSIXError 93 -> 100093
> ```
> 
> System logs captured while connecting micro:bit: [connect221.txt](https://github.com/ARMmbed/DAPLink/files/9881797/connect221.txt), [connect13B.txt](https://github.com/ARMmbed/DAPLink/files/9881798/connect13B.txt)
> 
> I experimented with **cp** in the terminal, and see
> 
> ```
> cp: microbit-testq.hex: could not copy extended attributes to /Volumes/MICROBIT/microbit-testq.hex: Attribute not found
> ```
> 
> **cp** shows the message above, does some flashing (perhaps not enough) and micro:bit eventually shows 504. **cp -X** does the same, but without the message. I used **xattr -c** the clear the extended attributes for a file. With that file, both **cp** and **cp -X** do the same with no message and trigger error 504.

> Here's a log from macOS Catalina, for connecting a V2.21. I edited out a lot of messages, most of which seemed to be informing all interested parties of the new disc. [catalina221.txt](https://github.com/ARMmbed/DAPLink/files/9881850/catalina221.txt)

martinwork, 2022-10-27:
Thanks @microbit-carlos. I was trying to figure out which bits of what I saw were still useful after the new info from @jaustin and yourself!

microbit-carlos, 2022-10-27:
No problem, thanks for capturing the data!

I've got a couple of USB wireshark captures as well, haven't had a chance to look into them yet, but leaving here in case somebody wants to give it a go:
- [cp-waitforunmount-remount.pcapng.zip](https://github.com/ARMmbed/DAPLink/files/9882075/cp-waitforunmount-remount.pcapng.zip)
    - This is running a `cp file.hex /Volumes/MICROBIT`, after about 12 seconds it doesn't transfer any more file data.
    - Usually at this point DAPLink is meant to unmount and remount the `MICROBIT` drive, but it doesn't do that until the end of the capture around the 30 seconds mark more or less.
- [drag-n-drop-only.pcapng.zip](https://github.com/ARMmbed/DAPLink/files/9882088/drag-n-drop-only.pcapng.zip)
     - This is using finder to drag and drop a hex file into the `MICROBIT` drive, so about 6 seconds of data.
     - I suspect the interesting bits will be right in the middle, as most of the time in the capture is me dragging the file on macOS and then navigating to the wireshark window to stop the capture

The micro:bit USB address would be `226.x.x`, so you can use the `usb.src matches "226"` filter to see only the relevant data.

manpowre, 2022-10-31:
anyone found a solution to this ? I just got the same issue with rp2040 chips. cant move the .uf2 firmware over :( reports as zero bytes file.

microbit-carlos, 2022-10-31:
You should be able to use a terminal and the `cp -X` command, or `rsync`.

jaustin, 2022-10-31:
Here's the micro:bit knowledge base workaround article. https://support.microbit.org/support/solutions/articles/19000139995-finder-error-100093-when-using-drag-and-drop-to-transfer-a-micro-bit-hex-file 

I'm about to add a section explaining how administrators of managed Apple devices can delay updates, too.

jaustin, 2022-11-01:
Similar to @aallan's workaround in https://github.com/raspberrypi/pico-sdk/issues/1081, I've been using a short automator script for micro:bit work - not at all sanitised for arbitrary use, but is safe enough for people to use themselves in their own trusted environment. It gives you an app you can drop a hex/UF2 file onto. Works nicely on the dock as a drop target...

Create an Automator project that's an "Application", then configure it like this screenshot (use `/Volumes/MICROBIT/` for a micro:bit device or `/Volumes/MBED` for older mbed devices)

[image]

(For RP2040 devices I think you'd want the following but I haven't got one to hand right now to test...)

[image]

kanthanp, 2022-11-10:
Platform: Apple MacBook Pro (2020 model) with macOS Ventura 13.0.1 (22A400)

First tested drag and drop with the NXP FRDM-K64F board via the mac's **Finder** tool. The K64F gives me the same error reported by @jaustin - `The operation can’t be completed because an unexpected error occurred (error code 100093)`. Like @willlordarm the binary appears to have flashed to the board when first checked but if you wait long enough the device will auto reboot and then when you check the content of the board the flashed binary is replaced with `failed.txt` which mentions that the transfer didn't complete. The binary I had flashed was the Blinky example program and I could see that it wasn't successful as the blinking LED on the K64F hadn't updated to the colour I had specified.

Then tested the deploy operation of Blinky example program a few times to the same K64F board using the **Mbed Studio** IDE. It performed mass storage device flashing (essentially drag & drop) of the binary successfully. The LED on the K64F had updated to the specified colour each time. 

Also prior to updating the MacBook to macOS Ventura it had macOS Monterey 12.4. I tested drag and drop with the same K64F board and Blinky example program via the mac's **Finder** tool. With this version of macOS the binary was successfully flashed and the LED colour on the K64F updated appropriately.

So for me the problem appears to be specifically with mac's **Finder** tool using **macOS Ventura**.

microbit-carlos, 2022-11-14:
We instrumented a DAPLink image with [RTT](https://www.segger.com/products/debug-probes/j-link/technology/about-real-time-transfer/) logging to capture the USB MSD reads and writes and have identified a couple of things relevant to this issue.

We should first clarify that we have two separate problems:
1) Drag and drop of a file into the MICROBIT drive aborts immediately in finder with 100093 error
2) As a workaround to the finder issue, using `cp` normally works, however even with `cp` there is a different issue flashing a file larger than 1 MB
    - `rsync` does work with files larger than 1 MB

Some of the content below makes references to a log file attached to the References section at the bottom of this response.

## Finder 100093 error

When a bin or a hex file are copied to the DAPLink `MICROBIT` drive, the data is processed and flashed into the target microcontroller as it is received, and the file contents themselves are discarded. For writes of any other file type, nothing is stored, and basically all writes are ignored. (There are also special files with the `.act` extension to trigger especial DAPLink operations, but these are not relevant to this issue).

We’ve confirmed with reasonable confidence that this error is triggered because DAPLink does not store the file data written to the MSD, and some metadata files created by macOS are now read back in Ventura.

In previous macOS releases (we are mostly testing on Monterey and Ventura), at the USB level we never see reads to these metadata blocks that had been written - we only see writes. Although we don't know the exact reason, we suspect the data was likely read from the OS disk cache, instead of reading directly from the drive, but don’t know how to verify that.

In Ventura, when looking at the logs from a failed attempt to drag and drop, we see reads to blocks that contain the hex file metadata (for example, we’ve identified at least one `._*` filename, more info in the Reference[1]), which we haven’t stored, so can’t return. After that there is no further valid data from the hex file written to the drive, and the 100093 error is raised.

We run a test by modifying DAPLink to cache in RAM some of this metadata being written and to return it when read by the OS, and in this test we can successfully flash hex files under 1 MB to the micro:bit with drag-and-drop from finder.

However, we cannot successfully write hex files greater than 1MB with finder nor `cp` either, with this test firmware or the default firmware.

## Finder/cp issue with files larger than 1 MB

For hex files larger than 1 MB neither `cp` (with unmodified DAPLink or the test-metadata-cacheing DAPLink) nor finder work in Ventura to successfully write programs that run on the micro:bit.

When we use either method to copy the data to the `MICROBIT` drive it appears to work initially but it ultimately times out. When we read back the flash contents of the target device, we see that after a certain point the device flash is blank, i.e. only part of the file has been written.

The point at which the data in the target flash stops corresponds to exactly 1MB of data transferred over USB (which for Intel Hex files 1 MB corresponds to around 375 KB of target data, as iHex is ASCII encoded, 16 bytes of data take 44 bytes of ASCII).

In our logs, the last hex file block of ASCII data written to the MSD [2] ends with `"0000000000000DE\n:10E240000000000055A803008975040000000000CC\n:10E"`

If we count the number of characters up to that point in the hex file, it is `1048576`, which is 1 MB exactly.

The following write immediately after this block is a change to the (root) directory table, to modify the properties of the table entry for the ihex file being written (in our case `intel-v2.hex`). Reference [3] shows the contents of this write.

After that, we see reads from one of the two metadata blocks followed by some writes [4].

After this the directory table entry for the `._*` file relating to the `intel-v2.hex` we are writing is modified to [5]

Finally, as far as useful work is concerned, we see multiple reads of the sector containing 
```
|........Mac OS X        ........|
|.2..............................|
|....................ATTR........|
…etc [6]
```
*The rest of the hex file is not written after this*, so it’s no surprise that the program does not run.

We also are not sure what’s happening at sector 1 and 2, 66 and 67 [7]

## A test that didn't work: Adding "no metadata" files

We created a test DAPlink firmware that creates these files in the mass storage drive, to hopefully stop macOS from creating some of the metadata files:
- `.fseventsd/no_data`
- `.Trashes`
- `.metadata_never_index`

With these files present, the `100093` error was still being triggered. The `.fseventsd/no_data` stopped the `.fseventsd/uuid` file from being created, but that didn't help resolve the issue.

## References:

RTT was used to capture log data via J-link debugger. We used a prototype micro:bit with an interface chip with a lot more RAM (128 KBs vs the usual 32 KB), so to load this hex file a micro:bit V2.2x with an nRF52833 interface chip is needed, this hex file won't work in any other micro:bit type, even if they also V2.20/V2.21: [nrf52820_microbit_if_crc.hex.zip](https://github.com/ARMmbed/DAPLink/files/10007294/nrf52820_microbit_if_crc.hex.zip)

This log file corresponds to a unsuccessful finder drag&drop of an Intel hex file named `intel-v2.hex`.

[... cut ...]
L09: 9100920093009400 9500960097009800 99009a009b009c00 9d009e009f00a000
L10: a100a200a300a400 a500a600a700a800 a900aa00ab00ac00 ad00ae00af00b000
L11: b100b200b300b400 b500b600b700b800 b900ba00bb00bc00 bd00be00bf00c000
L12: c100c200c300c400 c500c600c700c800 c900ca00cb00cc00 cd00ce00cf00d000
L13: d100d200d300d400 d500d600d700d800 d900da00db00dc00 dd00de00df00e000
L14: e100e200e300e400 e500e600e700e800 e900ea00eb00ec00 ed00ee00ef00f000
L15: f100f200f300f400 f500f600f700f800 f900fa00fb00fc00 fd00fe00ff000001
```

lurch, 2022-11-14:
Have you also tried `cp -X ...`, as https://www.raspberrypi.com/news/the-ventura-problem/ says that this then skips copying metadata?

microbit-carlos, 2022-11-15:
Yes `cp -X` has the same issue with files over 1 MB, and `rsync` does work. Although we haven't made a log capture of `rsync` to see how it differs, it'd be interesting to see how many writes/reads of the metadata files there are with `rsync` in comparison.

aallan, 2022-11-15:
I've added a comment to both [our blog post](https://www.raspberrypi.com/news/the-ventura-problem/) and our Github issue, raspberrypi/pico-sdk#1081, to this effect.

microbit-carlos, 2022-11-16:
A bit more info from Segger (J-Link) about the > 1 MB files: https://wiki.segger.com/J-Link_drag_and_drop_programming#macOS_specifics

lurch, 2022-11-16:
> A bit more info from Segger (J-Link) about the > 1 MB files: https://wiki.segger.com/J-Link_drag_and_drop_programming#macOS_specifics

Ah, good find. That says: "Apple macOS has a scheduler that splits data files writes to the drag & drop drive to 1 MB chunks that are delegated to multiple threads. This may lead to data being written to the drive non-sequentially because each 1 MB job of each thread may be interrupted by another thread at any time."
So AFAIK that would only affect `hex` and `bin` files, but it _wouldn't_ affect `uf2` files (as used on RP2040), as the [UF2 format](https://github.com/microsoft/uf2) was specifically designed to be able to handle problems like that.

lurch, 2022-12-06:
If you can't guarantee that the GUI or the OS-level `cp` command delivers data to the disk "in order" (because it's using multi-threaded copying, split into 1MB chunks), and you can't use UF2, I guess you'd need to rely on some custom program (e.g. a python script) to write the file-data to the disk in the order that you expect it? :man_shrugging:

microbit-carlos, 2022-12-06:
Yeah, thankfully at the moment using `rsync` works as a workaround, but it's not really something we can expect a large portion of the micro:bit userbase to be able to comfortably do.

Implementing something like UF2 in DAPLink would work, but that means all macOS users would have to update their micro:bit firmware, which the majority of users don't know anything about, and I'd go to the extent to say that it is something that they really shouldn't have to know about. The beauty of DAPLink is that it "just works" everywhere without the user needing to know it's there, and with the Ventura 1MB out-of-order changes it just fails without providing any useful information to the user as to what happened or how they can fix it.
Apart from that it also affects any other MSD-based tools that consume any kind of data file format that is not UF2, like Intel Hex, bin, srec, etc.

aallan, 2022-12-15:
Drag-and-drop of UF2 files for Raspberry Pi Pico and other RP2040-based boards has been fixed as of macOS Ventura 13.1. The fix does not resolve the issue where .hex files >1MB fail to write as file blocks are still out of order.

laa-star, 2023-02-02:
So, I upgraded to my Mac M1 to Ventura 13.2 about a week before getting my first PicoW. Maybe that was a mistake, because I have all the same uf2 flashing problems described with Ventura before 13.1. Or maybe I'm missing something because I'm new to the pico ecosystem. Can anyone confirm that the uf2 problem has not returned in Ventura 13.2?

aallan, 2023-02-03:
> Can anyone confirm that the uf2 problem has not returned in Ventura 13.2?

I can confirm that the UF2 flashing problem is **not** present in Ventura 13.2.

laa-star, 2023-02-03:
> > Can anyone confirm that the uf2 problem has not returned in Ventura 13.2?
> 
> I can confirm that the UF2 flashing problem is **not** present in Ventura 13.2.

Thank you for that. I did get it working. My actual problem was something else… I was following bad directions to make the on-board LED blink, and when it didn't work, I misdiagnosed the problem.

microbit-carlos, 2023-02-06:
Oh yes, thanks for the reminder @jaustin!

And thanks to Apple for fixing  the file>1MB issue! With the recent macOS Ventura 13.2 release this issue has been fully resolved and DAPLink should be working again. 🥳

closed 2023-02-06 by microbit-carlos

samskiter, 2023-09-11:
I *think* this has regressed. I managed to flash using picotool but DnD on 13.5.1 (22G90) wasn't working for me

microbit-carlos, 2023-09-21:
We've tested DAPLink with micro:bit on macOS 13.5.2 and had the occasional error, but not often enough to indicate this issue has regressed.
How common is the issue for you?

aallan, 2023-09-21:
Can confirm. We have seen no reoccurrence of this problem with the latest version of MacOS on the RP2040.