ARMmbed/DAPLink #987: Single DnD for multiple devices which have same flash algo programming  blob

ozersa, 2022-11-23:
We have some devices which has exactly same flash controller,  the changes between device to device are:

1. Flash start addr
2. Flash size
3. Flash sector size
4. RAM Size

As a sample [MAX32650](https://github.com/ARMmbed/DAPLink/blob/main/source/family/maxim/max32650/target.c)
[Maxim Devices](https://github.com/ARMmbed/DAPLink/tree/main/source/family/maxim)

The flash programming blob is exactly same for the devices. due to above differentiation we generate DnD for each device.

Is there a way to handle this delta in DAPLink FW and get one single DAPLink FW which supports DnD for multiple devices (the device that has same flash algo programming blob)?

Thanks

ozersa, 2022-11-23:
@mathias-arm  @flit

mathias-arm, 2022-11-23:
The support for Nordic DK boards detects (see [mkit_dk_dongle_nrf5x.c](https://github.com/ARMmbed/DAPLink/blob/main/source/board/mkit_dk_dongle_nrf5x.c) or [nrf5x_dk_ext.c](https://github.com/ARMmbed/DAPLink/blob/main/source/board/nrf5x_dk_ext.c)) at startup the target. It also show how the `target_cfg` can be set at start-up. In the Nordic case this is done with GPIO configuration, but you could also use SWD commands to detect the MCU version.

ozersa, 2022-11-24:
Thanks @mathias-arm , it will be good start point.
I took a quick look, it seems the configuration is done during power up [here](https://github.com/ARMmbed/DAPLink/blob/main/source/daplink/interface/main_interface.c#L302). That means the debugger requires to be connected to target board during power up or there should be on board debugger on target board.
In our case we have some boards that does not have on board debugger...

Just recap the purpose is to centralize DnD feature for multiple board. So that the ideal case would be detecting target device (by reading some register over swd...) at the beginning of each DnD process.

Does current DAPLink fw support this use case?

mathias-arm, 2022-11-24:
> I took a quick look, it seems the configuration is done during power up [here](https://github.com/ARMmbed/DAPLink/blob/main/source/daplink/interface/main_interface.c#L302). That means the debugger [...]

If the probe is not integrated with the target, It could introduce some challenges in terms of powering the probe and target in the right order (if the target can be powered through the Cortex-M debug connector, there might be some ways to reduce that uncertainty).

> Just recap the purpose is to centralize DnD feature for multiple board. So that the ideal case would be detecting target device (by reading some register over swd...) at the beginning of each DnD process.
> 
> Does current DAPLink fw support this use case?

There are no boards that currently do this AFAIK. But there are successful precedents for doing dynamic configuration at startup (Nordic DK boards are an example, Microbit v2 with `nrf52820` has support for different pin assignment). DAPLink does have the machinery to issue SWD commands (see [swd_host.c](https://github.com/ARMmbed/DAPLink/blob/main/source/daplink/interface/swd_host.c)).

closed 2023-02-04 by mbrossard