ARMmbed/DAPLink #988: stm32f103xb stack at bootloader

luckyxiaoer, 2022-11-25:
Hi, I tried to use Nucleo-L476RG board to port daplink using keil, but failed to get it work 

I've tried servial things 
- drag-n-drop
- flash hex file by merge bl.hex with if.hex 
- breakpoint debug

# drag-n-drop
I've generated stm32f103xb_bl  and stm32f103xb_stm32l476rg_if project,
after compiled them, I got bin and hex file .

I download stm32f103xb_bl to board using Jlink. 
and get a Maintenance disk in My computor. 

after I drag-n- droped stm32f103xb_stm32l476rg_if_crc.bin

the Maintenance disk disappered , but shown again.

my detial file in Maintenance disk show below.

```
# DAPLink Firmware - see https://daplink.io
Build ID: v0257-10-gc7b31bca (armcc, local mods)
Unique ID: 000000000671ff554852707267104749a5a5a5a597969908
HIC ID: 97969908
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Incompatible image detection: 0
Page erasing: 0
Daplink Mode: Bootloader
Bootloader Version: 0258
Git SHA: c7b31bca34819e9e406dfc5eb3506d03a898f495
Local Mods: 1
USB Interfaces: MSD
Bootloader CRC: 0xd37bb143
Interface CRC: 0x4915d882
Remount count: 0
URL: https://daplink.io
```
# flash merge file 
I use hexview to merge the bl.hex and if.hex
but the program still stuck in bootloader 

# breakpoint debug 
I debuged it in keil step by step , and find it failed to jump to app because of the 
```C
validate_bin_nvic((uint8_t *)g_board_info.target_cfg->flash_regions[0].start
```
and it returns false because 
```C
g_target_family && g_target_family->validate_bin_nvic
```
is always NULL 

and I found no function to refresh g_target_family  in bootloader .

# no idea
by now , I have no idea to solve this problem , and looking for your help , thank you ~

mathias-arm, 2022-11-25:
It's normal `g_target_family && g_target_family->validate_bin_nvic` would be false, it should move to the defautl validation function `validate_bin_nvic_base()`. You can check that the interface image starts with interrupt vectors that look consistent.
- Have you tried removing the call to `validate_bin_nvic()`?
- Have you tried building with GCC?
- Are you trying to port to NUCLEO-L476RG?

luckyxiaoer, 2022-11-27:
Hi mathias:
    it turns out to be the fault of the hardware config of the jumper5 in my NUCLEO-L476RG;
the Jump5 was lost before, so there was no power supply for stm32l476
and the NRST pin got no pull-up, and it  stay in low voltage level, even the reset bottom had not been pushed , so the program stuck in bootloader.

   after insert the jumper5， it shows a DAPLink disk in my computer .
and I compiled a gpio program for stm32l476 from the stm32 official demo. 

but it failed to be flashed into stm32l476 , and shows a  ASSERT.txt

```
Assert
File: ..\..\..\source\daplink\HardFault_Handler.c
Line: 67
Source: Application
Hexdumps
a44150b4
200023ec
20002fe8
00000000
20003008
08013a39
0801364a
21000024
40000000
00008200
00000000
00000000
a44150b0
a44150b0
a44150b4
200023ec
```

the ASSERT fault was caused by the wire connection for SWDIO 和 SWCLK between STM32L476 and STM32F103.
program runs normally, after wires removed.

closed 2022-11-27 by luckyxiaoer

mathias-arm, 2022-11-27:
Thank you for the update