ARMmbed/DAPLink #989: LPC11U35/401 support (OKdo E1)

sszczep, 2022-11-26:
Hey,

I have recently purchased OKdo E1 boards with LPC11U35/401 HIC and LPC55S69 target. It is visible under `lsusb` as `1fc9:0132 NXP Semiconductors LPC11U3x CMSIS-DAP v1.0.7`. Unfortunately, I couldn't get it to work with JLink nor pyOCD (I assume it's due to old CMSIS-DAP). The only thing I found working was MCUxpresso's `crt_emu_cm_redlink` tool. My goal is to use Zephyr and make `west flash`/`west debug` commands work. I guess DAPLink would make it possible, however it doesn't support this HIC due to missing RAM2 memory (#809). 

I would try remapping the memory layout, however it is too small (m_data + m_data_2 exceeds 8KB):
```Memory region         Used Size  Region Size  %age Used
    m_interrupts:          1 KB         1 KB    100.00%
          m_text:       54400 B        59 KB     90.04%
        m_cfgrom:          10 B         4 KB      0.24%
          m_data:        7124 B       7904 B     90.13%
        m_cfgram:         142 B        256 B     55.47%
        m_data_2:          2 KB         2 KB    100.00%
       m_usb_ram:          40 B         2 KB      1.95%
   text    data     bss     dec     hex filename
  55416    1052    8316   64784    fd10 build/lpc11u35_if.elf
  55416    1052    8316   64784    fd10 (TOTALS)
```

Can I somehow reduce memory usage/trim functionality to make it work? I most definitely do not need DnD and other things I'm not even aware of. If not, do I have other options rather than buying other board/soldering other HIC? 

I'm sorry if I'm missing something, it's my first time using debug probes. Any help appreciated.

mbrossard, 2022-11-27:
Hi @sszczep, I also have the OKdo E1. I have tried to fit DAPLink with the LPC55S69, but as you noticed the biggest challenge with supporting the LPC11U35/401 is losing that RAM2 region. I don't know how close I got, but it was not enough.

I have made an attempt to build DAPLink without DnD for the Okdo E1 in the [feature/okdo_e1](https://github.com/mbrossard/DAPLink/tree/feature/okdo_e1) branch. I have not tested it (I need to hunt the board), but removing the drive emulation and flashing functionality saved a lot of RAM and Flash.

```
Memory region         Used Size  Region Size  %age Used
    m_interrupts:          1 KB         1 KB    100.00%
          m_text:       28748 B        59 KB     47.58%
        m_cfgrom:          10 B         4 KB      0.24%
          m_data:        4400 B       7904 B     55.67%
        m_cfgram:         142 B        256 B     55.47%
       m_usb_ram:          40 B         2 KB      1.95%
   text	   data	    bss	    dec	    hex	filename
  29764	   1008	   3592	  34364	   863c	build/lpc11u35_okdo_e1_if.elf
  29764	   1008	   3592	  34364	   863c	(TOTALS)
```
Let me know if this works or if you can get it to work with some changes. It will need some cleanups to be mergeable.

sszczep, 2022-11-27:
Wow, haven't expected such a fast reply. So i flashed the CRC firmware, it gets detected in `lsusb`: `0d28:0204 NXP ARM mbed`. It also gets picked up by `pyocd`:
```
sszczep@sebastian-kde-neon:~$ pyocd list
  #   Probe/Board             Unique ID                                      Target  
-------------------------------------------------------------------------------------
  0   Arm DAPLink CMSIS-DAP   0000124a094c00000000000000000000000097969902   n/a     
      Okdo                    E1                                          
```
Certain `pyocd` commands such as reset or erase do not work (`0000694 E Transfer error while reading AHB-AP#1 ROM table: Memory transfer fault @ 0xe00fefd0-0xe00fefff [ap]`), however, I can successfully flash the image: 
```
sszczep@sebastian-kde-neon:~/zephyrproject$ source .venv/bin/activate
(.venv) sszczep@sebastian-kde-neon:~/zephyrproject$ cd zephyr
(.venv) sszczep@sebastian-kde-neon:~/zephyrproject/zephyr$ west flash -r pyocd
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner pyocd
WARNING: runners.pyocd: hex file (None) does not exist; falling back on .bin (/home/sszczep/zephyrproject/zephyr/build/zephyr/zephyr.bin). Consider enabling CONFIG_BUILD_OUTPUT_HEX.
-- runners.pyocd: Flashing file: /home/sszczep/zephyrproject/zephyr/build/zephyr/zephyr.bin
0000915 I Loading /home/sszczep/zephyrproject/zephyr/build/zephyr/zephyr.bin at 0x10000000 [load_cmd]
[==================================================] 100%
0002345 I Erased 32768 bytes (1 sector), programmed 15872 bytes (31 pages), skipped 0 bytes (0 pages) at 10.84 kB/s [loader]
```
Could not get debug to run (no logs after flashing samples/hello_world, might be a misconfiguration on my side). Also JLink nor MCUxpresso detects the probe. Is JLink support something achievable? Great job nonetheless.

mbrossard, 2022-11-27:
I was pretty aggressive and chose to use `hic_lpc11u35_bulk` that does not have the HID end-points. This can affect the compatibility with some tools. You can try to replace with `hic_lpc11u35` or  add `records/usb/usb-hid.yaml` (the LPC11U35 has a limited number of end-points, so we typically cannot have bulk and HID at the same time, but we can in this case since we don't have MSC).

I am wondering also if would make sense to add in `source/board/okdo_e1.c`:
```
    .board_id = "0236",
```
Which is the board ID for the LPC55S69-EVK board so some tools might be able to infer this a LPC55S69 target.

Another option would be adding `.target_cfg = &target_device,` to `g_board_info` and adding before it:
```
#include "target_config.h"

target_cfg_t target_device = {
    .version                        = kTargetConfigVersion,
    .target_vendor                  = "NXP",
    .target_part_number             = "LPC55S69JBD100",
};
```

@flit, any suggestions regarding pyocd issues?

mbrossard, 2022-12-31:
I updated [feature/okdo_e1](https://github.com/mbrossard/DAPLink/tree/feature/okdo_e1), with some other RAM savings changes.

```
Memory region         Used Size  Region Size  %age Used
    m_interrupts:          1 KB         1 KB    100.00%
          m_text:       53232 B        59 KB     88.11%
        m_cfgrom:          10 B         4 KB      0.24%
          m_data:        7856 B       7904 B     99.39%
        m_cfgram:         142 B        256 B     55.47%
       m_usb_ram:          40 B         2 KB      1.95%
arm-none-eabi-size --totals build/lpc11u35_okdo_e1_if.elf
   text    data     bss     dec     hex filename
  54248    1344    6712   62304    f360 build/lpc11u35_okdo_e1_if.elf
  54248    1344    6712   62304    f360 (TOTALS)
```

 This way I did not have to remove MSC and added the LPC55S69 target. Drag-n-drop is not working and there is an `ASSERT.TXT`, but `pyocd` picks up the target automatically:

```
  #   Probe/Board             Unique ID                                      Target
-----------------------------------------------------------------------------------------------
  0   Arm DAPLink CMSIS-DAP   0b06124913e600000000000000000000000097969902   ✔︎ lpc55s69jbd100
      Okdo                    E1
```

@flit, when I try to flash from the CMSIS Pack target, it fails:
```
(local) $ pyocd flash LPCXpresso55S69.hex
0000982 W Board ID 0b06 is not recognized [mbed_board]
0001849 I Loading LPCXpresso55S69.hex [load_cmd]
0001861 W Failed to add data chunk: memory region at address 0x00000000 is not writable [file_programmer]
0001861 I Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 0 bytes (0 pages) at 0.00 kB/s [loader]
```

But when I use the internal LPC55S69 target it works:
```
$ pyocd flash -t lpc55s69 LPCXpresso55S69.hex
0000981 W Board ID 0b06 is not recognized [mbed_board]
0001343 I Loading LPCXpresso55S69.hex [load_cmd]
[==================================================] 100%
0002703 I Erased 32768 bytes (1 sector), programmed 14848 bytes (29 pages), skipped 0 bytes (0 pages) at 10.76 kB/s [loader]
```

sszczep, 2022-12-31:
That's a massive update. Unfortunately I had to abandon my project for the time being, but I will make sure to check it out.

Thanks!

sszczep, 2023-01-05:
Hey @mbrossard, I've tried building the new version, however I can't get it to work - m_data exceeds 8000b. Have you used different flags by any chance? I simply cloned your fork and changed the branch to feature/okdo_e1.

mbrossard, 2023-01-08:
It was very close for me as well. I guess one possible explanation could I was using a newer GCC version?

```
$ arm-none-eabi-gcc -v
[...]
gcc version 10.3.1 20210621 (release) (GNU Arm Embedded Toolchain 10.3-2021.07)
```

sszczep, 2023-01-08:
Yes, that's what I've come up with. After upgrading to the latest version (12.2 I guess?) the size shrinked significantly. I had issues with undefined references to functions such as _write, _read, but was able to compile the project disabling fatal warnings in linker. Pyocd detects the lpc55s69jbd100 so it seems to work. As you mentioned previously, I think it should be detected as lpc55s69 instead, not to manually specify the target. I couldn't get it to work with MCUxpresso but I think we are on the right track