ARMmbed/DAPLink #1001: .yaml file tag that used to define include path for assembly (Asm) files.

ozersa, 2023-01-31:
In .yaml file I need to set include folder for assembly files (see below screenshot), but I do not know which tag need to be used for this purpose. 
The below ones does not work, could you let me know which tag use for this purpose?

[image]

mathias-arm, 2023-01-31:
Have you tried something like:
```
tool_specific:
    uvision:
        misc:
            asm_flags:
                - xxx
```

I think [m48ssidae.yaml](https://github.com/ARMmbed/DAPLink/blob/main/records/hic_hal/m48ssidae.yaml#L26) is doing what you are looking to do.

ozersa, 2023-02-01:
Thanks it works.
The PR related to this change [PR#1002](https://github.com/ARMmbed/DAPLink/pull/1002).

closed 2023-02-01 by mathias-arm