ARMmbed/DAPLink #1003: ld error when compiled with make_gcc_arm

IOsetting, 2023-02-01:
Toolchain version: 
arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi and arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi . 
The errors are the same.

Commands:
```bash
python tools/progen_compile.py -t make_gcc_arm --clean -v --parallel stm32f103xb_if
```
or
```bash
progen generate -t make_gcc_arm -p stm32f103xb_bl
make -C projectfiles/make_gcc_arm/stm32f103xb_bl
```

Compilation failed when linking, the console output:
```
Compiling ../../../source/usb/msc/usbd_core_msc.c
Compiling ../../../source/usb/msc/usbd_msc.c
Compiling ../../../source/usb/usbd_core.c
Assembling ../../../source/hic_hal/stm32/stm32f103xb/gcc/startup_stm32f103xb.S
Linking build/stm32f103xb_bl.elf
/opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-closer.o): in function `_close_r':
/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/closer.c:47: warning: _close is not implemented and will always fail
/opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-lseekr.o): in function `_lseek_r':
/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/lseekr.c:49: warning: _lseek is not implemented and will always fail
/opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-readr.o): in function `_read_r':
/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/readr.c:49: warning: _read is not implemented and will always fail
/opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-writer.o): in function `_write_r':
/data/jenkins/workspace/GNU-toolchain/arm-11/src/newlib-cygwin/newlib/libc/reent/writer.c:49: warning: _write is not implemented and will always fail
Memory region         Used Size  Region Size  %age Used
    m_interrupts:          1 KB         1 KB    100.00%
          m_text:         46 KB        46 KB    100.00%
        m_cfgrom:          0 GB         1 KB      0.00%
          m_data:        9168 B      20224 B     45.33%
        m_cfgram:         142 B        256 B     55.47%
collect2: error: ld returned 1 exit status
make: *** [Makefile:283: build/stm32f103xb_bl.elf] Error 1
make: Leaving directory '/home/milton/WorkC/DAPLink/projectfiles/make_gcc_arm/stm32f103xb_bl'
```

mbrossard, 2023-02-01:
This is a known issue. Using the 10.3 version of the compiler would be the recommended way for now.

If you want to test new compilers, add `-Wl,--no-warn-rwx-segment` to `ld_flags` in `records/tools/gcc_arm.yaml` and add the following lines in a `.c` file (e.g. source/daplink/sdk_stub.c):
```
void _close_r(void) {}
void _lseek_r(void) {}
void _read_r(void) {}
void _write_r(void) {}
```

IOsetting, 2023-02-02:
10.3 works. Thanks very much for your help!

closed 2023-02-02 by IOsetting