ARMmbed/DAPLink #1004: lpc55s69_mculink: USB descriptors appear to get corrupted

cstrahan, 2023-02-03:
USB descriptors appear to get corrupted after several accesses. 

I used `0257_lpc55s69_mculink_0x10000.hex` (from this release: https://github.com/ARMmbed/DAPLink/releases/tag/v0257) to program my MCU-LINK with DAPLink.

When I initially plug in my MCU-LINK, the descriptors usual look normal:

```
$ lsusb -vd '0d28:0204'

Bus 003 Device 038: ID 0d28:0204 NXP ARM mbed
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0d28 NXP
  idProduct          0x0204 ARM mbed
  bcdDevice           10.00
  iManufacturer           1 Arm
  iProduct                2 DAPLink CMSIS-DAP
  iSerial                 3 0000008084c40d1c5ed12d31085d16b94673bfa64c504355
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x008b
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               4 mbed Serial Port
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              4 mbed Serial Port
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          2
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
[... 314 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
cannot read device status, Resource temporarily unavailable (11)
```

I can reproduce this with two different computers, and two different MCU-LINKs.

I initially discovered this while trying to figure out why `pyocd` would initially discover the probe, and then subsequently fail: https://github.com/pyocd/pyOCD/issues/1497

jsiverskog, 2023-02-09:
also experiencing this, following.

jsiverskog, 2023-02-09:
attaching a totalphase beagle dump, it contains a full successful session, and then `pyocd list` which fails, and `lsusb -v` that fails.

failing `lsusb` starts at index 814513.

[usb.csv.zip](https://github.com/ARMmbed/DAPLink/files/10697810/usb.csv.zip)
[usb.tdc.zip](https://github.com/ARMmbed/DAPLink/files/10697815/usb.tdc.zip)

mathias-arm, 2023-02-09:
Could you try disabling high-speed by setting `USBD_HS_ENABLE` to `0` in `source/hic_hal/nxp/lpc55xx/usb_config.c`?

I have been looking for places where `USBD_ConfigDescriptor` and/or `USBD_ConfigDescriptor_HS` might get corrupted and two are related to high-speed (`USB_CONFIGURATION_DESCRIPTOR_TYPE` and `USB_OTHER_SPEED_CONFIG_DESCRIPTOR_TYPE`).

jsiverskog, 2023-02-10:
@mathias-arm : it fails to enumerate with that change, i've tried on different computers, dmesg:
```
[81915.843499] usb 3-4.3.2: unable to read config index 0 descriptor/start: -32
[81915.843511] usb 3-4.3.2: chopping to 0 config(s)
[81915.843515] usb 3-4.3.2: can't read configurations, error -32
[81915.926028] usb 3-4.3.2: new high-speed USB device number 88 using xhci_hcd
[81916.027072] usb 3-4.3.2: unable to read config index 0 descriptor/start: -32
[81916.027077] usb 3-4.3.2: chopping to 0 config(s)
[81916.027078] usb 3-4.3.2: can't read configurations, error -32
[81916.027573] usb 3-4.3-port2: attempt power cycle
[81917.541940] usb 3-4.3.2: new high-speed USB device number 90 using xhci_hcd
[81917.646794] usb 3-4.3.2: unable to read config index 0 descriptor/start: -32
[81917.646809] usb 3-4.3.2: chopping to 0 config(s)
[81917.646813] usb 3-4.3.2: can't read configurations, error -32
[81917.726003] usb 3-4.3.2: new high-speed USB device number 91 using xhci_hcd
[81917.831203] usb 3-4.3.2: unable to read config index 0 descriptor/start: -32
[81917.831226] usb 3-4.3.2: chopping to 0 config(s)
[81917.831230] usb 3-4.3.2: can't read configurations, error -32
```
tried with different hubs, and directly in the computer.

elfmimi, 2023-02-13:
I'm now preparing to validate my testimony below.
It goes, lpc55s69_bl/_if work fine with MCU-Link board. not lpc55s69_mculink_bl/_if .
At least that was the case when I tried it last time.

mathias-arm, 2023-02-14:
@cstrahan / @jsiverskog can you check or confirm @elfmimi's input?

jsiverskog, 2023-02-15:
@mathias-arm : i won't be able to test until next week.

cstrahan, 2023-02-17:
I'm trying @elfmimi's [suggestion](https://github.com/ARMmbed/DAPLink/issues/1004#issuecomment-1427184010, ).

Used `lpc55s69_bl_crc.hex` as the bootloader (instead of `lpc55s69_mculink_bl_crc.hex`) and `0257_lpc55s69__0x10000.bin` (instead of `0257_lpc55s69_mculink_0x10000.bin`).

Initial tests:

Before flashing:

```
$ probe-rs-cli list
The following devices were found:
[0]: DAPLink CMSIS-DAP (VID: 0d28, PID: 0204, Serial: 000000808bb7918e5d0379760db6608a0f9f6b9e4c504355, CmsisDap)
```

After:

```
$ probe-rs-cli list
The following devices were found:
[0]: LPC55xx DAPLink CMSIS-DAP (VID: 0d28, PID: 0204, Serial: 000000808bb7918e5d0379760db6608a0f9f6b9e4c504355, CmsisDap)
```

Will report what I find.

cstrahan, 2023-02-17:
I'm finding that the USB descriptors still get messed up:

```
$ lsusb -vd '0d28:0204'

Bus 003 Device 082: ID 0d28:0204 NXP ARM mbed
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0d28 NXP
  idProduct          0x0204 ARM mbed
  bcdDevice           10.00
  iManufacturer           1 Arm
  iProduct                2 LPC55xx DAPLink CMSIS-DAP
  iSerial                 3 000000808bb7918e5d0379760db6608a0f9f6b9e4c504355
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x008b
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               4 (error)
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              4 (error)
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          2
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
[... 91 lines cut ...]
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              9 (error)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
cannot read device status, Resource temporarily unavailable (11)
```

cstrahan, 2023-02-17:
I'm not entirely wedded to the MCU-LINK -- I only purchased it because, from rummaging around through the respective HIC and related header files, I found that it:

* Supports High Speed USB
* Supports JTAG
* It's $15 price is cheap enough for me to buy a couple

Though I don't know how much of a performance difference the High Speed makes in practice.

Until this issue is resolved, could anyone here make a recommendation for a similarly spec'd device with solid DAPLink support ?

In the interim, I'm using the latest proprietary MCU-LINK firmware, but I love the idea of using an open source solution, and I'm interested in seeing if DAPLink's support for flashing via MSC could be leveraged to achieve faster flashing (and thus quicker iteration during my software development). Faster flashing is my No. 1 desire right now :).

cstrahan, 2023-02-17:
I'll also note that the firmware otherwise (at least _seems_ to) run fine. After the descriptors go wonky, `pyocd` can no longer detect the probe (as the python usb lib fails-fast when trying to determine the `product` string if it can't ascertain the LANGIDs that the device supports), the usb lib used by `probe-rs` appears to default to assuming 0x0409 (English) is supported and thus determines the product string successfully, allowing it to find the probe.

There are cases where I need to use `pyocd`, though, and knowing that the device is already misbehaving in one known way makes me uneasy about depending on it during dev.

mbrossard, 2023-02-17:
I am on a Mac but I installed the port of `lsusb` and the output is stable as far as I can see:

```
Bus 020 Device 005: ID 0d28:0204  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 
  bDeviceSubClass         2 
  bDeviceProtocol         1 
  bMaxPacketSize0        64
  idVendor           0x0d28 
  idProduct          0x0204 
  bcdDevice           10.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x008b
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 
      bFunctionSubClass       2 
      bFunctionProtocol       1 
      iFunction               4 mbed Serial Port
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 
      bInterfaceSubClass      2 
      bInterfaceProtocol      1 
      iInterface              4 mbed Serial Port
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          2
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
[... 146 lines cut ...]

```diff
--- a/projects.yaml
+++ b/projects.yaml
@@ -27,7 +28,7 @@ module:
         - records/usb/usb-core.yaml
         - records/usb/usb-msc.yaml
         - records/usb/usb-cdc.yaml
-        - records/usb/usb-webusb.yaml
+        # - records/usb/usb-webusb.yaml
         - records/usb/usb-winusb.yaml
         - records/daplink/cmsis-dap.yaml
         - records/daplink/drag-n-drop.yaml
@@ -75,7 +76,7 @@ module:
         - records/rtos/rtos-cm33.yaml
         - records/hic_hal/lpc55s69.yaml
         - records/usb/usb-bulk.yaml
-        - records/usb/usb-hid.yaml
+        # - records/usb/usb-hid.yaml
     hic_m48ssidae: &module_hic_m48ssidae
         - records/rtos/rtos-cm4.yaml
         - records/hic_hal/m48ssidae.yaml
```

I am going to have to try to reproduce this on a Linux system.

mbrossard, 2023-02-17:
> I'm not entirely wedded to the MCU-LINK -- I only purchased it because, from rummaging around through the respective HIC and related header files, I found that it:
> 
> * Supports High Speed USB
> * Supports JTAG
> * Its $15 price is cheap enough for me to buy a couple
[...] 
> Until this issue is resolved, could anyone here make a recommendation for a similarly spec'd device with solid DAPLink support ?

`lpc55xx` is the only HIC supported in DAPLink that has both High-Speed and JTAG support. There is one probe called nanoDAP-HS based on the `sam3u2c` HIC running a modified DAPLink that adds JTAG support. I bought a couple (It's available on AliExpress for ~17$ in either USB-A or compact USB-C) with the hope of adding support in DAPLink.

> Though I don't know how much of a performance difference the High Speed makes in practice.
[...]
> I'm interested in seeing if DAPLink's support for flashing via MSC could be leveraged to achieve faster flashing 

Flashing through MSC should be faster in general, and High-Speed should not matter much through MSC (because Full-Speed is usually faster than MCU embedded Flash).

jsiverskog, 2023-02-22:
hello,
disabling webusb and hid seems to reduce this error occurrence significantly for us, but it's still happening now and then.

also, as far as i know, we've only seen this on linux, never on mac. perhaps the daplink is confusing the linux usb stack somehow? or if the linux usb stack is triggering something in the daplink.

mbrossard, 2023-02-25:
I now have a Linux machine (Ubuntu 22.04LTS) I have been able to partially reproduce. If I leave the MCU-Link plugged in at boot (also sometimes power-cycling the probe), I can see sometimes in `dmesg` logs:
```
 usb 3-2.2: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice=10.00
 usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
 usb 3-2.2: Product: DAPLink CMSIS-DAP
 usb 1-1.2: can't set config #1, error -71
```
I have seen also `error -110`.

But if that happens power-cycling (I have it on a USB hub with a per-port button):

```
 usb 3-2.2: new high-speed USB device number 4 using xhci_hcd
 usb 3-2.2: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice=10.00
 usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
 usb 3-2.2: Product: DAPLink CMSIS-DAP
 usb 3-2.2: Manufacturer: Arm
 usb 3-2.2: SerialNumber: 00000080a6d1d6915e5c1fe529c58fa238ffe6a64c504355
 cdc_acm 3-2.2:1.0: ttyACM0: USB ACM device
```

At first I had some issues reproducing the issue (maybe luck). I installed the udev rules (https://github.com/pyocd/pyOCD/tree/main/udev) and I was able to see the issue more often. It's has been pretty rare to have `lsusb` work and then seconds later not work. It might point to `udev` or something else in the Linux stack having a delayed behavior, things happening in the wrong order or interfering with each other. When I did observe a brief transient working state this is what I observed:

```
--- good.txt	2023-02-24 21:06:18.645504913 -0600
+++ bad.txt	2023-02-24 21:08:33.287803758 -0600
@@ -33,7 +33,7 @@
       bFunctionClass          2 Communications
       bFunctionSubClass       2 Abstract (modem)
       bFunctionProtocol       1 AT-commands (v.25ter)
-      iFunction               4 mbed Serial Port
+      iFunction               4
     Interface Descriptor:
       bLength                 9
       bDescriptorType         4
@@ -43,7 +43,7 @@
       bInterfaceClass         2 Communications
       bInterfaceSubClass      2 Abstract (modem)
       bInterfaceProtocol      1 AT-commands (v.25ter)
-      iInterface              4 mbed Serial Port
+      iInterface              4
       CDC Header:
         bcdCDC               1.10
       CDC Call Management:
@@ -77,7 +77,7 @@
       bInterfaceClass        10 CDC Data
       bInterfaceSubClass      0
       bInterfaceProtocol      0
-      iInterface              5 mbed Serial Port
+      iInterface              5
       Endpoint Descriptor:
         bLength                 7
         bDescriptorType         5
@@ -107,7 +107,7 @@
       bInterfaceClass       255 Vendor Specific Class
       bInterfaceSubClass      0
       bInterfaceProtocol      0
-      iInterface              6 CMSIS-DAP v2
+      iInterface              6
       Endpoint Descriptor:
         bLength                 7
         bDescriptorType         5
@@ -128,24 +128,3 @@
           Usage Type               Data
         wMaxPacketSize     0x0200  1x 512 bytes
         bInterval               0
-Binary Object Store Descriptor:
-  bLength                 5
-  bDescriptorType        15
-  wTotalLength       0x0021
-  bNumDeviceCaps          1
-  Platform Device Capability:
-    bLength                28
-    bDescriptorType        16
-    bDevCapabilityType      5
-    bReserved               0
-    PlatformCapabilityUUID    {d8dd60df-4589-4cc7-9cd2-659d9e648a9f}
-    CapabilityData[0]    0x00
-    CapabilityData[1]    0x00
-    CapabilityData[2]    0x03
-    CapabilityData[3]    0x06
-    CapabilityData[4]    0xaa
-    CapabilityData[5]    0x00
-    CapabilityData[6]    0x20
-    CapabilityData[7]    0x00
-Device Status:     0x0000
-  (Bus Powered)
```

jsiverskog, 2023-02-28:
@mbrossard that is a good first step to be able to resolve this, will you troubleshoot it further?

mbrossard, 2023-03-02:
I am borrowing a HS-capable USB analyzer to see if I can capture the difference between success and failure cases. I also need to check if I can reproduce with other HICs.

jsiverskog, 2023-03-03:
@mbrossard : cool, please keep us posted! thanks.

mbrossard, 2023-03-05:
With MCU-Link firmware, I have not been able to reproduce the issue as described. I see an issue if I boot the host (I am using a second computer to capture) with the MCU-Link plugged in. If I re-plug the MCU-Link everything seems to be working. I have captured over periods as long as 15 minutes, I have tried re-plugging multiple times.

As mentioned earlier I am using the latest Ubuntu (22.04). I am connecting the MCU-Link directly (no USB hub). I did try with a USB hub, I did not have any failures but it made the USB traffic much noisier.

@cstrahan, @jsiverskog: can you tell me a little bit more about your setup and steps to reproduce?

jsiverskog, 2023-03-10:
actually, you can also trigger it by running `pyocd load HEXFILE --target TARGET` while having the `lsusb` loop active.

mbrossard, 2023-03-11:
I was able to reproduce just with `watch -n0.5 sudo lsusb -v -d 0d28:0204` for something like 10 to 30 seconds. I added some debug logs and at this point it seems the probe simply stops receiving/processing USB interrupts. There does not seem to be a single pattern or obvious reason.

raveslave, 2023-03-20:
hi, I'm also very interested in getting a solution to this. I believe nxp must have solved this in their "offline branch" as I believe they make use of daplink with their mculink boards.

mbrossard, 2023-03-24:
I was able to reproduce the issue with USB Analyzer and debug probe connected with RTT logs from the LPC55. Unfortunately there does not a clear sequence of event that seems to cause the issue, nor is there an obvious crash of the software. The LPC55 simply seems to stop receiving USB traffic.

I have been looking at NXP UM11126 document, figure 164 and 165 in particular, to try to understand the driver. I have reached out to the original author of the driver (@flit), because I think we might not be one of the note:
`It is very important that the DevStatus(Setup) is only cleared after setting EP0OUT(Active), EP0OUT(Stall), EP0IN(Active) and EP0IN(Stall) bits are set to zero`.

cstrahan, 2023-06-16:
Anyone here know any engineers that work for NXP who might be able to shed some light on this? :slightly_smiling_face:

elfmimi, 2024-02-17:
I found a way to reproduce this efficiently in my environment.
I'm using slightly old Linux system. 
Debian 10.7 , kernel version: Debian 4.19.160-2 (2020-11-28)

while `while true; do lsusb -v -d 0d28:0204 >/dev/null; done` does trigger some times, it's rare.
on the other hand `lsusb -v -d 0d28:0204 > /dev/null & lsusb -v -d 0d28:0204 > /dev/null &`  triggers almost always. with or without use of a USB Hub.

elfmimi, 2024-02-18:
So, what's happening is as simple as SETUP bit of DEVCMDSTAT left high and never getting treated.
Externally resetting the SETUP bit by operating through, another, debug probe will return it to normal operation.

PR #1067 (2024-02-25): Fix LPC55 HIC missing out on a setup packet
Fix #1004
closed 2024-02-25 by mathias-arm